ARM-software/CMSIS_5 #211: // comment in CMSIS Core

CanastraRF, 2017-06-30:
In cmsis_armclang.h you add some //lint comment.

But // is not complient to older C Compiler.
Please replace 
//lint -esym(9058, IRQn)   disable MISRA 2012 Rule 2.4 for IRQn
with
/*lint -esym(9058, IRQn)*/   /*disable MISRA 2012 Rule 2.4 for IRQn*/

In cmsis_compiler.h and cmsis_gcc.h are some comments with //
Please replace them with /* ... */

JonatanAntoni, 2017-06-30:
Hi CanastraRF,

Thank you for your detailed code review.

We are typically relying on C99 standard, thus // (C++-style comments) should be fine as well.
The compiler specific header file `cmsis_armclang.h` is intended to be used with ARM Compiler 6, which has C99 support. In GCC support for Cortex controllers was added much later than C99 support.

Can you give me an example which old (non-C99) compiler you would like to see supported? Or do you have other reasons for compiling your code with C89 mode?

Kind regards,
Jonatan

CanastraRF, 2017-06-30:
Hi Jonatan
There are some projects with C90 as e requirement.
So we configure the code analyze (PC-Lint) tool to C90.
Then we must modify cmsis_* to comply the requirements.

JonatanAntoni, 2017-06-30:
Hi @CanastraRF,

would you mind to share the changes you need as a pull request?

Kind regards,
Jonatan

CanastraRF, 2017-06-30:
Hi Jonatan
Sorry im not familar with a pull request.
But I attached my modification.
[cmsis.zip](https://github.com/ARM-software/CMSIS_5/files/1114764/cmsis.zip)
Kind regards
Reto Felix

JonatanAntoni, 2017-07-13:
Hi @CanastraRF,

I changed to comment style on devel branch now.

May I ask you to review if the changes works for you, please?
Please consider closing this issue if you do not have any questions left.

Cheers,
Jonatan

CanastraRF, 2017-07-18:
Hi @Jonatan
Your modification is not correct.
PC-Lint search for //lint .. or /*lint ..*/
If there is any space between //* and lint it is not recognized.
Also explanation of the lint comment must be in a separate comment.

Examble:
/* lint -esym(9058, IRQn)   disable MISRA 2012 Rule 2.4 for IRQn */
should be replaced with
/*lint -esym(9058, IRQn)*/   /* disable MISRA 2012 Rule 2.4 for IRQn */

Please fix this
Reto Felix

JonatanAntoni, 2017-07-19:
Hi @CanastraRF,

Thank you for making me aware.
I fixed the comments in f97808b.

Cheers,
Jonatan

CanastraRF, 2017-07-19:
Thank you
Cheers,
Reto Felix

closed 2017-07-19 by CanastraRF