ARM-software/CMSIS_5 #215: SCS (System control space registers) not in any SVD?

cs2dsb, 2017-07-15:
I can't find anywhere that the SCS registers (System control, MPU etc.) are documented in a program readable format.

As these are Arm peripherals included in the same locations on all devices I expected them to be documented in an Arm svd if not documented in the vendor ones.

The registers I'm talking about are documented in ARMv7-M Architecture Reference Manual (Issue E.b) - Chapter B3.

Thanks

JonatanAntoni, 2017-07-18:
Hi @cs2dsb,

currently we do not have those core peripherals as a part of SVD files. In MDK the core peripherals are handled differently and not using the generic system viewer feed by SVD files.

May I ask you to be a little more specific about the requirements you have, please? What do you want to use the core peripheral information for? Which level of detail do you need?

From firmware project point of view the core peripherals are well defined in the generic core header files. Beside some effort to do so it should be possible to put the definitions to an SVD file as well.

Cheers,
Jonatan

cs2dsb, 2017-07-18:
Hi @JonatanAntoni,

I'm using https://github.com/japaric/svd2rust to generate rust code for accessing the peripheral registers. 

It's a work in progress but it's working great for the peripherals defined in the vendor SVD files and I was hoping to do the same for the SCS peripherals.

The tool is using the peripheral name, description, base address and address block, then for each register in the peripherals it's using register name, description, address offset, size and reset value and for the fields in the register it's using field name, description, bit offset, bit width, access and then either a write constraint or a list of enumerated values to indicate the possible field values.

Most of the vendor SVDs don't provide the write constraint or enumerated values so I've been adding them to a patch that gets applied to the SVD before running it through the tool. It's a shame the vendors don't provide the full detail but at least having the addresses and widths provided is a whole bunch of typos and copy/paste errors I don't have to make and fix!

It seems like CMSIS-SVD is a great tool for letting people use a wider variety of languages without Arm and the vendors having to support them directly and it would be great if the core peripherals were part of that if possible.

Thanks,
Daniel

JonatanAntoni, 2017-07-19:
Hi @cs2dsb,

Thank you for sharing some insights. Your approach sounds reasonable. 

We do something very similar for C header files but limited to the device specific peripherals. The core peripherals are common and hence implemented once in CMSIS core header files. Thus we never had the inducement to put them into an SVD file.

Would you mind to create some SVD files? Contributions are highly appreciated.

Cheers,
Jonatan

cs2dsb, 2017-07-23:
Hi @JonatanAntoni,

Sure, I'm currently adding the registers/fields as I need them but I'll make a task to go through and make a complete set when I've got some time later in the project.

I'll close this issue for now but I've created an issue in our private repo to contribute them back here once they are done.

Thanks,
Daniel

closed 2017-07-23 by cs2dsb

JonatanAntoni, 2017-07-24:
Thanks, Daniel.