ARM-software/CMSIS_5 #217: Add windowing functions to DSP library

escherstair, 2017-07-20:
CMSIS-DSP offers some functions to perform FFT (Fast Fourier Transform).

An usual approach, before a FFT, is to apply a window, for example to reduce leakage effect.

A lot of different window have been developed and tested in DSP, but common choice is between Hanning, Hamming, Blackman and Flat-Top (even if [many others](https://en.wikipedia.org/wiki/Window_function) are available).

I suggest the developers to add to CMSIS-DSP some functions to apply (at least) these common windows (f32, q15 and q31 data formats).

JonatanAntoni, 2017-07-20:
Hi @escherstair,

thank you for sharing your enhancement ideas. 

Would you be able to contribute in this area? Any input is highly appreciated.

Cheers,
Jonatan

escherstair, 2017-07-20:
Hi @JonatanAntoni,

I can share my experience if needed

jonnor, 2019-03-01:
I was in need of this now for some machine learning. So I added a small Python tool generate C code for a window function table to the emlearn library. After generating the table one can multiply it with input signal to apply the window.
https://github.com/jonnor/emlearn/tree/master/examples/window-function.py

```bash
pip install emlearn
emlearn-window-function --window hann --length 512 --name my_fft_window --out fft_window.h
```

```c
#include "fft_window.h"
...
arm_mult_f32(input, (float32_t *)my_fft_window, input, INPUT_LENGTH);
...
```

christophe0606, 2020-01-23:
I close the ticket since, as explained in previous comment, arm_mult can be used to multiply the signal with a given window and it is not the goal of the CMSIS-DSP to provide tables for all kind of windows and window length.

closed 2020-01-23 by christophe0606