ARM-software/CMSIS_5 #218: Typo in arm_var_q31() documentation?

escherstair, 2017-07-20:
The documentation for arm_var_q31() says:

> The function is implemented using an internal 64-bit accumulator. The input is represented in 1.31 format, which is then downshifted by 8 bits which yields 1.23, and intermediate multiplication yields a 2.46 format. The accumulator maintains full precision of the intermediate multiplication results, but provides **only a 16 guard bits**.

What does "only a 16 guard bits" mean?
I suspect some kind of typo

llefaucheur, 2017-11-10:
Hi escherstair, the Variance is computing a sum of squares with 8bits provision for overflows, which translates to 16bits margin once you squared the input. The "sumOfSquares" of the code has really 16bits margin against overflow.

escherstair, 2017-12-15:
Ok,
so I think

> The accumulator maintains full precision of the intermediate multiplication results, but provides only a 16 guard bits.

should be modified into

> The accumulator maintains full precision of the intermediate multiplication results **and so it** provides 16 guard bits.

Do you agree?

llefaucheur, 2017-12-15:
Accepted ! (English is not my mother language).

christophe0606, 2020-01-22:
Corrected in commit https://github.com/ARM-software/CMSIS_5/commit/aa5e7eb149750b1d503bc39e16812f985f2b8d4f

closed 2020-01-22 by christophe0606