ARM-software/CMSIS_5 #220: Output format for arm_rfft_q31

escherstair, 2017-07-21:
The documentation for `arm_rfft_q31` says

> Internally input is downscaled by 2 for every stage to avoid saturations inside CFFT/CIFFT process. Hence the output format is different for different RFFT sizes. The input and output formats for different RFFT sizes and number of bits to upscale are mentioned in the tables below for RFFT and RIFFT:
[image]
**Input and Output Formats for Q31 RFFT**

I'm not sure the table contains the right values.
If I run the following snippet of code
```c
#include "arm_math.h"

q31_t input[1024];
q31_t output[1024];
arm_rfft_instance_q31  	S;
arm_status stat;

/* fill the input[] buffer with four cycles of a sine wave with amplitude 0x3FFFFFFF */
for(int i=0; i<256; i++) {
    input[i] = arm_sin_q31 (0x00800000*i);
    input[i+256] = input[i];
    input[i+512] = input[i];
    input[i+768] = input[i];
}

/* init the instance structure */
stat = arm_rfft_init_q31 (&S, 1024, 0, 0);

/* calculate FFT of the input[] buffer
 * the expected output is the whole buffer set to 0, except for imag[4]
 * because the input[] buffer has been filled in with exactly four cycles of sine wave */
arm_rfft_q31 (&S, input, output);
```

I find that the amplitude of imag[4] if `0x1FFFFFFF` anf I can't upscale it by 9 bits.
Moreover, if I perform a FFT with size 512 (and so only two cycles of sine wave are included) I find the same aplitude `imag[2] = 0x1FFFFFFF`

Could you double check, please?

xizhizhang, 2017-08-23:
the documentation means arm_rfft_q31 will upscale 9 bits, when it calculates.

so, in order to get the result that you wish, all of your input should downscale 9 bits at first, before using arm_rfft_q31.
`
/* fill the input[] buffer with four cycles of a sine wave with amplitude 0x3FFFFFFF */
for(int i=0; i<256; i++) {
    input[i] = arm_sin_q31 (0x00800000*i) >> 9;
    input[i+256] = input[i] >> 9;
    input[i+512] = input[i] >> 9;
    input[i+768] = input[i] >> 9; 
}
`

escherstair, 2017-08-24:
Sorry, I don't understand.
The documentation says that `arm_rfft_q31` **downscales** internally and you write that it **upscales**.

Can you double check and clarify, please?

xizhizhang, 2017-08-24:
sorry, I am wrong. it is downscale.

I have tried to run your code, and got the result: imag[4] = 0xC0000002.(amplitude is -64)

by the way, I modified 
`arm_rfft_init_q31 (&S, 1024, 0, 0)`
to
`arm_rfft_init_q31 (&S, 1024, 0, 1)`

this amplitude is -64, and I checked it in python rfft implement, the result is OK.

And, do you actually want to get the inverse fft of input?
refers to your input, the inverse fft of them, amplitude is 0.5(also can represent as 0x3fffffff in Q31, just same with your code comment).

escherstair, 2017-08-25:
@xizhizhang I fixed two typos in my code and now I hope it's much more clear:
- output buffer should have a size of 2048 elements (see #219)
- the amplitude of the sine wave is `0x7FFFFFFF` (maximum positive value for `q31_t`)
- bitReverseFlag should be enabled in `arm_rfft_init_q31`
- I find an amplitude for FFT bin equal to `0x3FFFFFFF`
I'm interested in the direct FFT transform, which for a sine wave produces exactly one line with amplitude equal to the peak value of the wave (scaled in some well-documented way, if this is the case).

If you run the code you should find the following values in `output[]` buffer:
```
output[0] = real[0] = -16
output[1] = imag[0] = 0
output[2] = real[1] =0
output[3] = imag[1] =0
output[4] = real[2] =0
output[5] = imag[2] =0
output[6] = real[3] =0
output[7] = imag[3] =0
output[8] = real[4] =-6
output[9] = imag[4] =-1073741820	
output[10] = real[5] =0
...
```
You can see that the component [4] of FFT is almost imaginary (this depends on the phase of the sine wave), and its amplitude is `0x1FFFFFFF` (this is the amplitude of the bin of FFT transform; it's half the peak value of sine wave).
You can easily see that I can't 9-bits upscale `0x1FFFFFFF` without overflow.

My question refers to documentation that seems wrong.

xizhizhang, 2017-08-26:
@escherstair 
yeah, my result is also output[9] = imag[4] =-1073741820(0xc0000004).

then, refers to documentation 0xc0000004 is format in 10.22, which means, it is already a large negative number.

for example, to format Q31(1.31), 0xc00000004 equals to -0.5,  and to 2.30, 0xc0000004 equals to -1, to 10.22, 0xc0000004 equals to -256

in one word, the result need not upscale.

Maybe the documentation is wrong, but, I think the error is 9-bits->10-bits at most, not 9->0, 9->1, or 9->2, and so on.

escherstair, 2017-08-28:
@xizhizhang 
maybe the confusion is how to intepret the result.
If I calculate FFT transform of Q31(1.31) data, I would expect Q31 results.
Otherwise I'm not sure if the transform has sense: let me explain.
If I have a sine wave with amplitude 0.5, I know that its FFT transform has exactly one bin, with amplitude 0.5 (independently on which format I use to represent it).
Since I can express the sine wave in Q31 format, I expect that its FFT transform can be expressed in Q31 format too.

Probably the implementation of `arm_rfft_q31` doesn't involve the FFT normalization factor (which is `1/FFT_size`).
This is why the output from FFT is a big number and the output format is different depending on RFFT size.
I'm interested in the "normalized FFT", and probably the output from `arm_rfft_q31` needs normalization (quite a common need).

If this is the misunderstanding, I suggest to clearly add this info to the documentation:

> The output of `arm_rfft_q31` is NOT normalized.

xizhizhang, 2017-08-28:
@escherstair 
yeah, you are right, arm_rfft_q31 is not normalized in the source code.

By the way, rfft in numpy which is the python math library, is also not normalized in default. New added in version 1.10. Maybe add the normalized factor is a good idea.

escherstair, 2017-08-29:
If I'm right, you can interpret the output as it is now as "normalized FFT" with format Q30 (2.30) for every FFT length.

llefaucheur, 2017-11-10:
Hi escherstair, would you like more details, or can we close this topic ?
Kind regards, Laurent.

escherstair, 2017-12-15:
Hi @llefaucheur 
I think the situation is quite clear, but I suggest to improve the documentation a little bit.
It would be better if the following information is added:

- if you interpret the output from `arm_rfft_q31` as "not normalized", its output format is described in "Output Format" in the table (it is is different for different RFFT sizes). In this situation, no upscale is necessary.

- if you interpret the output from `arm_rfft_q31` as "normalized", its output format is Q30 (2.30) for every RFFT sizes

Do you agree this clarifies the situation?

fgr1986, 2018-11-19:
> Hi @llefaucheur
> I think the situation is quite clear, but I suggest to improve the documentation a little bit.
> It would be better if the following information is added:
>
>     * if you interpret the [...]

@escherstair ,  thanks for raise the issue.
As you point out, probably the documentation could be clearer.
And as @xizhizhang pointed, I think there may be an error on the internal downscaling, or at least on the documentation. It looks like during the fft computation, the downscaling factor reach 2**-9 for a 512 rfft operation, being the output in Q10.22, instead the documented 2**-8 and Q9.23.

@JonatanAntoni, @llefaucheur, could you please provide more light on the issue?

Thanks

llefaucheur, 2018-11-26:
The arm_rfft_q31 algorithm guarantees a full-scale input translates to full-scale output. For example a table of 64 samples at 3FFFFFFF will result in a full scale output at 3FFFFFE0, while it should be 64 times larger in floating points. The programmer is told the output is rescaled by log2(64) = 6bits. So the input Q1.31 is translated to Q7.25. (and not[ Q6.26](http://www.keil.com/pack/doc/CMSIS/DSP/html/group__RealFFT.html#gabaeab5646aeea9844e6d42ca8c73fe3a)) . This test was the opportunity to check the fft output of 64 x 7FFFFFFF, which is wrong.

caseymillsdavis, 2019-03-14:
I have a similar complaint as @escherstair. The documentation for the output of the rfft fixed point functions, both q15 and q31, appear to be incorrect or at least ambiguous. 

I hit this when testing arm_rfft_q15 on a 1-cycle with full-scale amplitude (value 32767 in q15_t). The magnitude of the value output in bin-1 is 16383. I'm using a vector length of 128 and so from the documentation I'm expecting the output to be in Q7.9. That's clearly not the case. It appears to be Q2.14.

If I use arm_cfft_q15 the output is simply Q1.15 and everything is as expected but I'd like to use real-input optimized version for the memory saving.

christophe0606, 2020-01-23:
It is now covered in issue https://github.com/ARM-software/CMSIS_5/issues/810

closed 2020-01-23 by christophe0606