ARM-software/CMSIS_5 #225: arm_std_f32.c can give negative input to arm_sqrt_f32, giving 0.0 erroneously

underpickled, 2017-08-04:
Function calculates the mean of squares (MoS) and square of means (SoM), and feeds (MoS - SoM) to arm_sqrt_f32(). This value can be negative, which causes arm_std_f32 to return 0.0 erroneously. 

Std Dev can never be negative because it's the square root of a squared number (variance). arm_std_f32 is built for speed, so the calculation is done a bit differently, but there should at least be an absolute value or other negative number guard before supplying the input to arm_sqrt_f32().

`/* Compute Mean of squares of the input samples
   * and then store the result in a temporary variable, meanOfSquares. */
  meanOfSquares = sumOfSquares / ((float32_t) blockSize - 1.0f);

  /* Compute mean of all input values */
  mean = sum / (float32_t) blockSize;

  /* Compute square of mean */
  squareOfMean = (mean * mean) * (((float32_t) blockSize) /
                                  ((float32_t) blockSize - 1.0f));

  /* Compute standard deviation and then store the result to the destination */
  arm_sqrt_f32((meanOfSquares - squareOfMean), pResult);`

underpickled, 2017-08-04:
Implementation Suggestion:

`if (meanOfSquares > squareOfMean)
{
    arm_sqrt_f32((meanOfSquares - squareOfMean), pResult);
}
else
{
    arm_sqrt_f32((squareOfMean - meanOfSquares), pResult);
}`

JonatanAntoni, 2017-08-07:
Thank you for this detailed feedback. We will look into it.

llefaucheur, 2017-09-28:
Standard deviation is the square-root of variance.
Variance can never be negative, because it is the average squared deviation from the mean.
The lowest variance comes from data series with no variation at all, and the result must be returned as zero.
Due to rounding effects in the floating point computations you may have a very small negative variance, which later is translated as zero after square-root.

llefaucheur, 2017-09-28:
Hi Dylan, to go further on the issue you observed, could please share with us the test pattern revealing the issue (MoS < SoM), and the content of the register FPSCR which holds the floating-point rounding conditions ? Kind regards, Laurent.

underpickled, 2017-09-28:
Hi Laurent,
I had considered that this could be a result of rounding, but since implementing my suggestion above on my own fork of CMSIS I see the problem occur far less often. Once in a while it pops up likely due to rounding error, but before making that change it was happening nearly every time. The input was raw data from a sensor on a product I'm developing, so there's not a dedicated test pattern that produces the result. If I have time soon I'll try to collect some data that will illustrate the problem.
Regards,
Dylan

llefaucheur, 2017-09-28:
Do you agree the variance must be always larger or equal to zero, so having the square-root subroutine returning zero should not be an issue, it simply tells 'this signal is just a DC offset'. If we cannot have the sequence of data, may be you can share the FPSCR content.

underpickled, 2017-09-28:
Yes, of course, the variance must always be greater than zero. However, the method of calculating the variance in arm_std_f32 is not the "textbook" method. Typically you take the mean of all the values, and then sum the squares of differences from the mean. In this approach the square of the mean is subtracted from the mean of the summed squares.

christophe0606, 2020-01-22:
I removed the label "bug" because the function is behaving as expected.
But it is a naive implementation of the variance algorithm and is not numerically stable.

When the arm_std_f32 is returning a negative value then there is no way to recover a meaningful result. 

The assumption made in this implementation is that the result is 0.
It may be justified because often negative result is generated when the true variance is small.

Anyway, it means this function will have to be replaced by a totally different and numerically stable implementation.

Use of this function may require to do some scaling of the data in input to avoid numerical instabilities and cancellations.

I'll create another issue to request a new numerically stable implementation of this function.

closed 2020-01-22 by christophe0606