ARM-software/CMSIS_5 #227: Misra violation in mpu_armv7.h, mpu_armv8.h

CanastraRF, 2017-08-17:
When I check mpu_armv7.h, mpu_armv8.h against Misra rules C++ 2008 I get some
Violation in the following lines:
``#define ARM_MPU_AP_NONE 0u 
#define ARM_MPU_AP_PRIV 1u
#define ARM_MPU_AP_URO  2u
#define ARM_MPU_AP_FULL 3u
#define ARM_MPU_AP_PRO  5u
#define ARM_MPU_AP_RO   6u
...
``
Rule 2-13-4 Literal suffixies shall be upper case
Please fix this soon
Reto Felix

CanastraRF, 2017-08-29:
Hi Jonatan,
Thank you for Fixing this issue in most places.
But PcLint reports the following issues:

`  if (data == 0u) { return 32u; }`
               ^
cmsis_compiler.h(138): Note 1960: Violates MISRA C++ 2008 Required Rule 2-13-4, Lower case literal suffix: u

`  if (data == 0u) { return 32u; }`
                             ^
cmsis_compiler.h(138): Note 1960: Violates MISRA C++ 2008 Required Rule 2-13-4, Lower case literal suffix: u

`    count += 1u;`
              ^
cmsis_compiler.h(145): Note 1960: Violates MISRA C++ 2008 Required Rule 2-13-4, Lower case literal suffix: u

`    mask = mask >> 1u;`
                    ^
cmsis_compiler.h(146): Note 1960: Violates MISRA C++ 2008 Required Rule 2-13-4, Lower case literal suffix: u

`  uint32_t mask = 0x80000000;`
                  ^
cmsis_compiler.h(141): Note 1960: Violates MISRA C++ 2008 Required Rule 2-13-3, Unsigned octal and hexadecimal literals require a 'U' suffix
cmsis_compiler.h(141): Note 9105: unsigned octal and hexadecimal literals require a 'U' suffix [MISRA C++ Rule 2-13-3]

`  while ((data & mask) == 0)`
         ~~~~~~~~~~~~~ ^  ~
cmsis_compiler.h(143): note 9029: an unsigned value and a signed value cannot be used together as operands to == [MISRA 2012 Rule 10.4, required]
cmsis_compiler.h(143): Note 1960: Violates MISRA C++ 2008 Required Rule 5-0-4, Implicit conversion changes signedness

mpu_armv7.h(118): Note 1917: empty prototype for definition, assumed '(void)'

mpu_armv8.h(59)
`#define ARM_MPU_ATTR(O, I) (((O & 0xFu) << 4U) | (((O & 0xFu) != 0U) ? (I & 0xFu) : ((I & 0x3U) << 2U)))`

mpu_armv8.h(168)
`  const uint32_t mask = 0xFFu << pos;`

Please fix them too.
Reto Felix

JonatanAntoni, 2017-08-29:
Hi @CanastraRF,

Thanks for double checking. I have looked over those. The latest commit should fix them as well.

Cheers,
Jonatan

CanastraRF, 2017-08-30:
Hi Jonatan
Thanks again.
Now it complain better to misra
Reto

closed 2017-08-30 by CanastraRF