ARM-software/CMSIS_5 #228: Round robin issue

AmberAussie, 2017-08-17:
Hi,

I have a issue that a task is never executed. I was able to reproduce it with 3 task:

 - TaskHighPrio: which use 50% of cpu ressources
 - TaskRoundRobin1: use as much time as possible
 - TaskRoundRobin2: use as much time as possible

The problem is that on rtx_system.c on function osRtxTick_Handler the robin.tick is reseted because half the time TickHandler come during TaskHighPrio and other half it come on TaskRoundRobin1. Then TaskRoundRobin2 is never started & executed.

I think round robin must be handled for each priority not just sampled on a tick basis.

JonatanAntoni, 2017-08-29:
Hi @AmberAussie,

Thank you for your report and the analysis. I was able to create a small reproducer that reveals the problem.

Currently we do not store round robin per priority. Its some kind of weighing up accuracy versus memory requirements.

An easy solution might be to put blocked threads behind all others of same priority instead of ahead. This would give all threads the chance to run but the actual time slices might not be equally distributed among all threads. I.e. if a round robin thread is preempted after using only part of its time slice the rest of the time slice is gone. Thus the other robin thread might actually get more execution time if it is less often preempted.

Do you have any concerns about such a "lazy" solution? A more elaborate solution would come at additional costs.

Kind regards,
Jonatan

AmberAussie, 2017-08-29:
Hi @JonatanAntoni ,

Thank you for your proposal, yes it would fix that particular problem but after reflection the fact that CPU is not be equally distributed is also a unexpected behavior on some other applications.

To achieve the correct scheduling, I don't see another solution than having a counter per priority or per thread or adding limitation (round robin between lowest priority task only or OS user shouldn't wake up of task < round robin time or ...)

I propose to add a robin_tick on os_thread_t that we initiate with osRtxInfo.thread.robin.timeout, decrements on osRtxTick_Handler until it reach 0. Once it reach 0 if there is another task with same priority ready we put the task back in thread list at the end of it's priority and reload robin_tick with osRtxInfo.thread.robin.timeout.

Or maybe something based on tick count between context switch... but it could be that it's often less than a tick so it's harder.

Kind regards

JonatanAntoni, 2017-08-29:
Thanks for sharing your thoughts. I will discuss the issue with the dev team and report back.

JonatanAntoni, 2017-09-01:
Hi @AmberAussie,

We discussed the issue in the team and apologize that the current implementation has some drawbacks.

Initially we never thought of priority and round robin scheduling to be used in combination. But I see the value to use this for some low priority and independent background tasks. 

We discussed to move the round robin tick count from RTX kernel control block to the thread control blocks. Thus each individual thread gets its own robin counter. This comes at an additional cost of 4 bytes per thread control block, i.e. for 10 threads 40 bytes of RAM. The code size and implementation is only adopted slightly which should not grow the code size.

Do you have any concerns about growing memory needs?

Cheers,
Jonatan

AmberAussie, 2017-09-05:
Hi,

No problem, all code has bugs. The memory is not a problem in our case.

Thank you

mishka88ru, 2020-07-22:
Hi @JonatanAntoni 
is https://github.com/ARM-software/CMSIS_5/blob/098cf8406f1463fc2cde3ee0ed8fecf4eba0690e/CMSIS/RTOS2/RTX/Include/rtx_os.h#L102 the correct data type to look into?

I'm just curious: is it possible to fix this bug using a counter? If a thread works for 500 ns and then gets preempted, then these 500 ns are lost (or rounded up to 1 full tick). I'm not an expert in cpu/rtos design... may be I am wrong...

I think, to achieve more or less "true" round robin, core should save systick value on each IRQ (may be on the stack), so RTOS later could calculate the amount of work.

JonatanAntoni, 2020-07-27:
Hi @mishka88ru,

Currently the round robin counter is stored in the kernel control block. In order to combine priority based scheduling with round robin, we'd need to to save the information per priority group. This needs proper thinking and design. 

And yes, all timing based on ticks in RTX5 is rounded. So, if a scheduling happens between two ticks the thread either looses some fractions of a tick or gains them. Some drawbacks will probably stay if combining both scheduling techniques.

Cheers,
Jonatan

RobertRostohar, 2021-05-17:
Should be resolved with the mentioned patch.

JonatanAntoni, 2021-06-08:
Hi @gzzi, may I ask you to check the fix and close the issue, please?

gzzi, 2021-06-11:
I do not have anymore access to the hw where issue was found, but it seem ok.

Thanks

closed 2021-06-11 by gzzi