ARM-software/CMSIS_5 #232: CMSIS-DSP DCT Type IV Transform functions

alkeer, 2017-08-24:
In the dct4 document with regards to length supported by the transform it is mentioned as below,

```
Lengths supported by the transform:
As DCT4 internally uses Real FFT, it supports all the lengths supported by arm_rfft_f32(). The library provides separate functions for Q15, Q31, and floating-point data types.
```
Ref: [http://www.keil.com/pack/doc/CMSIS/DSP/html/group__DCT4__IDCT4.html](url)

Also in the rfft function it is given as ,

`The parameter fftLen Specifies length of RFFT/CIFFT process. Supported FFT Lengths are 32, 64, 128, 256, 512, 1024, 2048, 4096`

Ref:[http://www.keil.com/pack/doc/CMSIS/DSP/html/group__RealFFT.html#ga3df1766d230532bc068fc4ed69d0fcdc](url)

So I assume **DCT type IV transform function also supports all the lengths 32, 64, 128, 256, 512, 1024, 2048, 4096 supported by rfft**. But in dct4 initialization function **arm_dct4_init_datatype()**  

> arm_dct4_init_f32.c, arm_dct4_init_q15.c, arm_dct4_init_q31.c

Twiddle & cosfactor values are only given for these four lengths **128, 512, 2048, 8192** . Hence I'm not able to use **arm_dct4_init_datatype()**  for any other FFT lengths. 

`datatype -> f32/q31/q15`

Eg ref: [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_dct4_init_f32.c](url)

**Does this mean dsp dct4 transform function can be applied only to these four lengths 128, 512, 2048, 8192?**

llefaucheur, 2017-10-31:
You are true. DCT4_F32 works for the lengths  128, 512, 2048, 8192.
Adding more lengths consists in increasing the options of the switch/case located in arm_dct4_init_f32() and defining the corresponding weights on top of the file arm_dct4_init_f32.c. There is formula to help computing the coefficients. Kind regards, Laurent.

alkeer, 2017-11-10:
@llefaucheur Thanks for your reply. It means that dct4 can be applied for all other lengths as well using the formula to calculate their coefficients. For simplification purpose,  in the  function arm_dct4_init_f32() they have defined for only four lengths.

closed 2017-12-04 by llefaucheur