ARM-software/CMSIS_5 #233: CMSIS-PACK missing ROM/RAM sections in pdsc file for Cortex A targets

TTornblom, 2017-08-29:
IAR EWARM extracts the memory sections information from the .pdsc XML file to build link scripts for the different targets.

For Cortex-M0 this information looks like:
---
```
      <memory id="IROM1"                                start="0x00000000" size="0x00040000" startup="1" default="1"/>
      <memory id="IRAM1"                                start="0x20000000" size="0x00020000" init   ="0" default="1"/>
```
---

This information is missing for all Cortex-A targets, which makes the creation of the link scripts silently fail, and linking then fails with a few undefined symbols, INTVEC, CSTACK and HEAP.

JonatanAntoni, 2017-08-29:
Hi @TTornblom,

thanks for letting us know. We are going to add this information to the PDSC file for Cortex-A devices as well, soon.

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
@TTornblom, May I ask you to cross check the solution and finally close this issue, please?

TTornblom, 2019-03-12:
Looks good, closing this issue

closed 2019-03-12 by TTornblom