ARM-software/CMSIS_5 #236: Make warning M371 into an INFO in SVDConv

David-Garcia-Polo, 2017-09-06:
SVDConv version 3.3.9 issue a warning in case of defined headerStructName is equal to the default headerStructName if not given.

Can this warning be made an Info?

JonatanAntoni, 2019-03-12:
Hi @David-Garcia-Polo,

May I ask you to check this with the latest version of SVDConv, please?
Let me know if the problem still exists for you.

Thanks,
Jonatan

David-Garcia-Polo, 2019-03-13:
will take a look

simtind, 2019-03-13:
Looks like we're still seeing the issue, running the attached [nrf52811.txt](https://github.com/ARM-software/CMSIS_5/files/2960655/nrf52811.txt) in SVDConv 3.3.25 with the command:
`SVDConv.exe nrf52811.txt --generate=header -x info` 
still warns:
> CMSIS-SVD SVD Consistency Checker / Header File Generator V3.3.25
> Copyright (C) 2010 - 2018 ARM Ltd and ARM Germany GmbH. All rights reserved.
>
> Arguments: "nrf52811.txt" --generate=header -x "info"
>
> *** WARNING M371: nrf52811.txt (Line 47)
>   Cluster 'FICR_INFO' <headerStructName> is equal to hirachical name
>
> *** WARNING M221: nrf52811.txt (Line 3)
>  Input filename must end with .svd: 'nrf52811.txt
>
>
> Found 0 Error(s) and 2 Warning(s).

JonatanAntoni, 2019-03-13:
Hi @simtind, 

Thanks for providing that reproducer. We need to take another look to it.

Is this a blocker for you? I.e. is the generated header what you expect? Or is it only about the criticality of that message?

Cheers,
Jonatan

simtind, 2019-03-13:
We're currently building header files with -x M371, and haven't seen any problems. 
The generated headers are as expected, so it is not critical, but having to build with ignored warnings is uncomfortable.

JonatanAntoni, 2020-01-21:
Hi @simtind,

Sorry for late reply. I revived the discussion.

The warning actually disappears if you simply remove the `<headerStructName>` completely, unless you intent to have a non-default name. If you want to keep it due to reasons we might really reduce the warning into an info message. I'll let you know about that after further discussions about possible side-effects.

Cheers,
Jonatan