ARM-software/CMSIS_5 #237: SVDConv marks % character as invalid with warning M216

David-Garcia-Polo, 2017-09-06:
% character is not an invalid character in xml and should therefore not be elimimnated from description fields.

simtind, 2017-12-13:
Issue can be tested against attached test file.

[test_percent_sign.zip](https://github.com/ARM-software/CMSIS_5/files/1554787/test_percent_sign.zip)

JonatanAntoni, 2017-12-13:
Thanks for providing the reproducer. We need to work on an enhancement here, because in general we use % sign to introduce variable substitution.

simtind, 2017-12-15:
Ok, is there something we can do to a percent sign to get it through your substitution process unchanged? 
\% or something?

JonatanAntoni, 2017-12-15:
Hi @simtind,

we are currently think about allowing '%%' to get a single '%' in the output. It will take us some time to bring this enhancement into the tool.

Best,
Jonatan

simtind, 2017-12-15:
Sounds good, thank you for the prompt feedback.