ARM-software/CMSIS_5 #238: Compilation with `-Wstrict-prototypes` produces warnings on GCC

FreddieChopin, 2017-09-08:
When the code using CMSIS 5 is compiled with GCC with added `-Wstrict-prototypes` following warnings are produced:
```
arm-none-eabi-gcc -Wall -Wstrict-prototypes -Wshadow -std=gnu99 -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -g -ggdb3 -O2  -ffunction-sections -fdata-sections  <a lot of include paths...> -c stm32f4xx_hal.c -o ../../../output/external/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.o
In file included from ../../distortos/external/CMSIS/core_cm4.h:1944:0,
                 from ../../distortos/external/CMSIS-STM32F4/stm32f407xx.h:183,
                 from ../../distortos/external/CMSIS-STM32F4/stm32f4xx.h:151,
                 from ../../distortos/source/chip/STM32/STM32F4/include/distortos/chip/CMSIS-proxy.h:69,
                 from ../Inc/stm32f4xx_hal_conf.h:42,
                 from ../Inc/stm32f4xx_hal.h:48,
                 from stm32f4xx_hal.c:54:
../../distortos/external/CMSIS/mpu_armv7.h:118:22: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 __STATIC_INLINE void ARM_MPU_Disable()
                      ^~~~~~~~~~~~~~~
```

Seems like an easy thing to fix with no downsides, so what do you think? I could provide a pull request if necessary.

FreddieChopin, 2017-09-08:
OK, I see this is already fixed in the code, so nevermind (;

Any idea when could we expect a new release?

closed 2017-09-08 by FreddieChopin

JonatanAntoni, 2017-09-08:
Hi @FreddieChopin,

thank you for making us aware of warnings we could work on.
I used this trigger to raise the warning level used by our validation tests.

Cheers,
Jonatan