ARM-software/CMSIS_5 #240: Clash with c11's noreturn definition in cmsis_os2.h

filiperinaldi, 2017-09-11:
When using C11's stdnoreturn.h header, noreturn is defined as _Noreturn. If an unit includes both stdnoreturn.h and cmsis_os2.h, the following code from cmsis_os2.h breaks the compilation:

```
#ifndef __NO_RETURN
#if   defined(__CC_ARM)
#define __NO_RETURN __declspec(noreturn)
#elif defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
#define __NO_RETURN __attribute__((noreturn))
#elif defined(__GNUC__)
#define __NO_RETURN __attribute__((noreturn))
#elif defined(__ICCARM__)
#define __NO_RETURN __noreturn
#else
#define __NO_RETURN
#endif
#endif
```
More specifically,  __attribute__((noreturn)) is expanded to  __attribute__((_Noreturn)) which is an invalid attribute.

Possible solutions:

1.
GCC allows adding underscores around the attribute name to avoid the such issues with macro names. Example:  __attribute__((__noreturn__))
The other compilers might have similar feature.
https://gcc.gnu.org/onlinedocs/gcc/Attribute-Syntax.html#Attribute-Syntax

2.
Detect the C version and use C11 _Noreturn keyword for compilers that support it.

RobertRostohar, 2017-10-30:
cmsis_os2.h has been updated to use `__noreturn__` for GCC and ARMCC6 (clang) as suggested.

@JonatanAntoni
We should update the "noreturn" also in CMSIS core headers.

JonatanAntoni, 2017-11-06:
Hi @filiperinaldi,

May I ask you to double check the changes made, please?

Thanks,
Jonatan

filiperinaldi, 2017-11-09:
Hi Jonatan,

I tested (with GCC) a recent version of the **develop** branch with the two patches above and they work fine.
Thanks,
Filipe

JonatanAntoni, 2017-11-09:
Thanks for the feedback. I'll close the issue.

closed 2017-11-09 by JonatanAntoni