ARM-software/CMSIS_5 #241: Build Error in arm_math.h when using IAR compiler with ARM_MATH_ARMV8MBL

kjassmann-renesas, 2017-09-12:
I'm using CMSIS 5.1.0.  I get the following errors when I try to build the ARM DSP library for CM23.  The problem is that these functions are already defined for IAR CM23.  They are redefined in arm_math.h if ARM_MATH_CM0_FAMILY is defined, and ARM_MATH_CM0_FAMILY is defined for ARMV8MBL in arm_math.h.

Build output:

   IAR ANSI C/C++ Compiler V8.11.3.13950/W32 for ARM
   Copyright 1999-2017 IAR Systems AB.
   Evaluation license - IAR Embedded Workbench for ARM, Evaluation Edition 8.11

    CMSIS_INLINE __STATIC_INLINE uint32_t __CLZ(
                                          ^
"<local_path>\arm_math.h",582  Error[Pe147]: declaration is incompatible with "__nounwind __interwork __softfp unsigned char __CLZ(unsigned long)" (declared at line 209 of "C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.0\arm\inc\c\intrinsics.h")

    CMSIS_INLINE __STATIC_INLINE q31_t __SSAT(
                                       ^
"<local_path>\arm_math.h",708  Error[Pe147]: declaration is incompatible with "__nounwind __interwork __softfp signed long __SSAT(unsigned long, unsigned int)" (declared at line 257 of "C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.0\arm\inc\c\intrinsics.h")

JonatanAntoni, 2017-09-13:
Hi @kjassmann-renesas,

I am sorry for this inconvenience. We has some unclear responsibilities in the past according the fallback implementations of those intrinsic functions. Currently we are updating all the cmsis_compiler.h stuff and address this issues as well. 

Is it okay for you to remove those duplicates locally for the time being? 
You might also try the latest development branch. Please let us know if you face further issues. 

Thank you very much for your valuable feedback.

Best regards,
Jonatan

kjassmann-renesas, 2017-09-13:
Thank you for fixing this.  We can definitely work around it for now.  Do you know when this update will be officially released? We would like to include CMSIS DSP support for CM23 in one of our future releases.  Thanks!

kjassmann-renesas, 2017-10-04:
This actually appears to be easier to fix by updating cmsis_iar.h, which is distributed by IAR.  We are working with IAR to get this done.  Please consider this resolved.

closed 2017-10-04 by kjassmann-renesas

JonatanAntoni, 2017-10-09:
Hi @kjassmann-renesas,

Please be aware that we are currently working on moving cmsis_iar.h to be part of CMSIS (renamed to cmsis_iccarm.h) and not be shipped with the compiler anymore. 

Cheers,
Jonatan