ARM-software/CMSIS_5 #242: obsolit condition in inline __USAT

CanastraRF, 2017-09-13:
Dear Support
With last update you add an inline implementation for __USAT() in cmsis_*.h like the following:

```
__attribute__((always_inline)) __STATIC_INLINE uint32_t __USAT(int32_t val, uint32_t sat)
{
  if (sat <= 31U) {
    const uint32_t max = ((1U << sat) - 1U);
    if (val > (int32_t)max) {
      return max;
    } else if (val < 0) {
      return 0U;
    }
  }
  return val;
}
```
As long as val is unsigned the condition if (val < 0) is always evaluated to false and can removed.

BTW add a Symbol 
`
#define __STATIC_INLINE_ALWAYS __attribute__((always_inline))  static inline
`
and use it for inline function like the above.

Reto Felix

JonatanAntoni, 2017-09-14:
Hi Reto,

yes, I added this fallback implementation for __USAT (and __SSAT) to the compiler headers to have them available on Cortex-M0 devices (which do not have the intrinsics in ARMv6-M).

We discussed the signature to be used (i.e. val int32_t or uint32_t). I tend to use the siganture as defined by the [Arm C Language Extensions](https://www.google.co.uk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjXsO6bkKTWAhVMCMAKHcszCyUQFggnMAA&url=http%3A%2F%2Finfocenter.arm.com%2Fhelp%2Ftopic%2Fcom.arm.doc.ihi0053d%2Findex.html&usg=AFQjCNFvBki1UOs36HxVt3--qDqteAaA4A).

In fact val is interpreted to be signed, i.e. input values lower then zero are saturated to zero. Thus the second condition is needed for the fallback implementation as well. You might want to take a look to the CoreValidation tests I recently added. This tests are used to validate that all the intrinsics defined in the core headers behave similarly on different architectures and compilers.

Cheers,
Jonatan

CanastraRF, 2017-09-15:
Hi Jonatan,
Thank you for your Explanation.
I agree, that the inline implementation should have the same siganture as defined by the Arm C Language Extensions.
So Your implemetation is well.
Cheers,
Reto Felix

closed 2017-09-15 by CanastraRF