ARM-software/CMSIS_5 #245: Hooks for thread terminate and exit

c1728p9, 2017-09-14:
One feature of mbed-os testing is to capture and report worst case stack usage. This works for threads that are still active at the end of the test, but does not capture information on the threads that have exited or have been terminated. A feature that would be beneficial to mbed-os is a thread exit and thread terminate hook, so stack usage can be recorded. As a workaround mbed-os is planning on using the event mechanism to record this information in https://github.com/ARMmbed/mbed-os/pull/4684.

JonatanAntoni, 2017-09-15:
Hi Russ,

We have discussed the issue and are currently investigating for a good solution.

Do you have any additional requirements for those hooks outside of pure testing purposes?

Typically we try to avoid adding features only for rare edge cases in order to keep RTX "simple and stupid" in favor of ease-of-use. So for testing purposes the event mechanism seams to be a valid solution. Do you see any drawbacks of this solution?

Cheers,
Jonatan

c1728p9, 2017-09-18:
There isn't a strong need for this outside of testing. The most useful case would be stack metrics for devices in the field. There is no requirement for this though.

The only downside of this approach is some RTX code needs to be duplicated and added to the handler to keep RTX Events working as expected. Any example of this can be seen  [in the change adding this to mbed-os](https://github.com/ARMmbed/mbed-os/pull/4684/commits/a2b53011bec2b39ddfb445b627488e98567a3bee#diff-6246b64fb25b1c4d97734e8cb593ecb6R154).

deepikabhavnani, 2017-10-30:
Hello Jonatan,
Any updates on this functionality?
Regards,
Deepika

JonatanAntoni, 2017-11-06:
Hi all,

we discussed the issue and think it should be solved quiet well with overwriting the event functions.
A really clean solution would be having hooks for all the event functions. On the other hand such a solution would be some kind of overwhelming. We want to make such an enhancement with caution and only if we see broader usage scenarios.

What do you think. Is the current solution acceptable for the time being?

Cheers,
Jonatan

deepikabhavnani, 2017-11-06:
@bulislaw - Please share your thoughts on suggested solution.

c1728p9, 2017-11-06:
@JonatanAntoni I think this is acceptable for the time being, although it is a bit of a hack. The biggest concern I have is that internal RTX event handling needs to be done in the code which overrides the event hook, otherwise uvision event logging won't work as expected.

https://github.com/ARM-software/CMSIS_5/blob/d20b8aad7f5e4188608ff0c26246cc6e4961b1ba/CMSIS/RTOS2/RTX/Source/rtx_evr.c#L794-L801

 A potentially low impact change that could be made, is remove weak from EvrRtx* and add a call to an empty weak function from it. That way the code in mbed-os can't break the event handling and doesn't need to worry about its internals. 

```c
#if (!defined(EVR_RTX_DISABLE) && (OS_EVR_THREAD != 0) && !defined(EVR_RTX_THREAD_EXIT_DISABLE))
__WEAK RtxThreadExitHook(void) { };
void EvrRtxThreadExit (void) {
#if defined(RTE_Compiler_EventRecorder)
  EventRecord2(EvtRtxThreadExit, 0U, 0U);
#else
#endif
  RtxThreadExitHook();
}
#endif
```

JonatanAntoni, 2017-11-07:
Hi @c1728p9,

exactly this solution was discussed here as well. But if we take this solution we would like to introduce those hooks commonly, not only for EvrRtxThreadExit. Thus the change would introduce a bunch of new functions which seems to be a bit overwhelming to us at the moment. We'd like to keep this enhancement idea on the backlog to be implemented if we see a broader demand.

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
@ReinhardKeil, @RobertRostohar: We should finally conclude on this one. Do we want to keep this as an enhancement or drop it?