ARM-software/CMSIS_5 #246: IAR: warnings with optimization disabled - svc always inline

0xc0170, 2017-09-22:
We got an issue reported here https://github.com/ARMmbed/mbed-os/issues/5117. You can find more details there. Tested with IAR 7.80

I'll paste here warnings:

```
[Warning] rtx_delay.c@77,4: [Go005]: Cannot inline function "__svcDelay" at this level of optimization
[Warning] rtx_delay.c@77,4: [Go004]: Could not inline function "__svcDelay" (inlined function is too complex or inlining is turned off)
[Warning] rtx_delay.c@87,4: [Go005]: Cannot inline function "__svcDelayUntil" at this level of optimization
[Warning] rtx_delay.c@87,0: [Go004]: Could not inline function "__svcDelayUntil" (inlined function is too complex or inlining is turned off)
Compile [ 58.9%]: rtx_evflags.c
[Warning] rtx_evflags.c@517,4: [Go005]: Cannot inline function "__svcEventFlagsNew" at this level of optimization
[Warning] rtx_evflags.c@517,4: [Go004]: Could not inline function "__svcEventFlagsNew" (inlined function is too complex or inlining is turned off)
[Warning] rtx_evflags.c@526,4: [Go005]: Cannot inline function "__svcEventFlagsGetName" at this level of optimization
[Warning] rtx_evflags.c@526,6: [Go004]: Could not inline function "__svcEventFlagsGetName" (inlined function is too complex or inlining is turned off)
```

It seems that IAR produces lot of warnings if optimizations are off (debug profile) for each SVC defined function, as they are defined with always inlined.

As an answer from IAR development team, the pragma would solve this for now (to supress warnings). I can propose a patch if this is acceptable.

We would like to eliminate these warnings

JonatanAntoni, 2017-09-22:
Hi @0xc0170,

Thanks for making us aware of this issue.

One can globally disable this warnings in IAR projects settings for the time being.

I would like to investigate further if this warnings are only nasty ones or point to a real issue.

Most parts of RTX are compiler agnostic. So we need to find a way to add those pragmas for IAR without cluttering the source code for all other compilers. 

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
@0xc0170 any updates on this? From RTX point of view we cannot do much about it. In later versions of IAR this should be solved. Please let us know if we need to further discuss this issue. Otherwise I'd like to ask you to finally close this issue, please.

TTornblom, 2019-03-12:
Add the following to Project -> Options -> C/C++ Compiler  -> Diagnostics -> Suppress these diagnostics:
---
Go004,Go005
---

or use the following option if compiling from CLI:
---
--diag_suppress Go004,Go005
---

ReinhardKeil, 2019-03-20:
Closing as the solution is provided

closed 2019-03-20 by ReinhardKeil