ARM-software/CMSIS_5 #248: Overuse of type casts in __NVIC_EnableIRQ() and related functions

franzhollerer, 2017-09-27:
I think there is an overuse of type casts in __NVIC_EnableIRQ() and related functions.

```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn)
{
  if ((int32_t)(IRQn) >= 0)
  {
    NVIC->ISER[(((uint32_t)(int32_t)IRQn) >> 5UL)] = (uint32_t)(1UL << (((uint32_t)(int32_t)IRQn) & 0x1FUL));
  }
}
```

If you look at the index into NVIC-ISER:

> NVIC->ISER[(((uint32_t)(int32_t)IRQn) >> 5UL)] = ...

I don't see why the expression

> (((uint32_t)(int32_t)IRQn) >> 5UL)

should give a different value than

> (((uint32_t)IRQn) >> 5UL)

Have I missed something?

JonatanAntoni, 2018-01-12:
Hi @franzhollerer,

May I ask you to double check the changes provided and close the issue if solved, please?

Thanks,
Jonatan

franzhollerer, 2018-01-12:
Hi @JonatanAntoni,

```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn)
{
  if ((int32_t)(IRQn) >= 0)
  {
    NVIC->ISER[(((uint32_t)IRQn) >> 5UL)] = (uint32_t)(1UL << (((uint32_t)IRQn) & 0x1FUL));
  }
}
```
Yeah, it is much more readable now. IMHO the code is correct. Thanks for the modification.

Here are some furhter suggestions, without having them tested myself. Also, I don't know if they are MISRO compliant. Thus you may like to cross-check them with your colleagues. Or keep the code as it is now.

*Further suggestions:*

As this is a function (and not a macro), I think there is no need to put IRQn in parentheses within the if-statement.

Furthermore, I would not cast IRQn to (uint32_t) for the left shift operation. The if-statement grants that it is positive. Furthermore, the compiler could issue a warning if it sees a negative constant, which will not work if it is casted to an unsigned value.

Thus, I would finaly come to:
```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn)
{
  if ((int32_t )IRQn >= 0)
  {
    NVIC->ISER[(((uint32_t)IRQn) >> 5UL)] = (uint32_t)(1UL << (IRQn & 0x1FUL));
  }
}
```

JonatanAntoni, 2018-01-12:
Hi @franzhollerer,

Thanks for your contribution, review and further suggestions.

The parenthesis in the if-clause can be omitted or placed like in the rest of the function, i.e `if (((int32_t )IRQn) >= 0)`. This is a minor issue we could change if touching the file again.

The cast for the bitwise operation (&) is required by MISRA C:2012 Rule 10.1 "Operands shall not be of an inappropriate essential type". The actual integer type used for an enum is implementation-defined. Furthermore bitwise operations on signed integer types are implementation-defined as well.

Best,
Jonatan

franzhollerer, 2018-01-12:
@JonatanAntoni 
I agree. If you have to follow MISRA then you have to keep the cast. I have no objection on that argument.

> Furthermore bitwise operations on signed integer types are implementation-defined as well.

I would like to make this statement more precise. From the C99 standard:

> If the value of the right operand is negative or is greater than or equal to the width of the promoted left operand, the behavior is undefined.

Being said, as long as IRQn is not negative the operation is well defined, regardless of whether it is a signed or unsigned integer type. And the if-statement ensures that it is non-negative.

Furthermore, the typical use case looks like this:

> __NVIC_EnableIRQ(UART2_IRQn);

The function is directly called with the enum value. If the enum is negative, the compiler can detect this at compile time and warn us if this results in undefined behavior. Our at least a good static analysis tool can to that.

The type cast prevents this. Concerned code keeps buggy, but the problem reveals sometime at runtime. The MISRA essential type rules do more harm than good.

But you have no choice anyway  :-(

JonatanAntoni, 2018-01-15:
Hi @franzhollerer,

thanks for clarifying.

The use case you described is the major one we have in mind as well.
Having this function declared as `NVIC_EnableIRQ(IRQn_Type irq)` it is totally fine for the compiler to use it like

> NVIC_EnableIRQ(SysTick_IRQn)

which of course if semantically wrong. The if-statement catches this. So there is no undefined behavior, is it? For sure the code stays buggy because the `NVIC_EnableIRQ` simply does nothing when used this way, i.e. the intended result (enabled System Tick interrupt) is not achieved.

It might be challenging to harden this part of the code in a compatible way. What do you think?

Best,
Jonatan

franzhollerer, 2018-01-15:
@JonatanAntoni 

The thing I wanted to point out (especially for people how follow this discussion) is that whith modern compilers and static code analysis tools, less type casts increase the chance to find programming errors in a early stage. While following the MISRA rules displaces the real problem.

But this was more or less an academic objection triggered by a reflex against some MISRA rules :-)

You have the if-statement which is fine. I don't see a serios option to further harden this part whihout breaking compatiblity. We cannont use static_assert() as there might be code which pass a variable whith its value determined at runtime to NVIC_EnableIRQ() instead of an enum constant known at compile time.

The code is fine. I will close the issue.

closed 2018-01-15 by franzhollerer

JonatanAntoni, 2018-01-15:
Hi @franzhollerer,

thanks for valuable feedback. 
I think blindly following MISRA is counterproductive. Using a cast for silencing MISRA checker should only be the last resort if being sure the code is correct. Too bad not to have another resort in this case.

Please keep pointing out code you think its worth refactoring.

Best,
Jonatan