ARM-software/CMSIS_5 #249: CMSIS SystemInit architecture fails. any update to RAM variables will be lost.

David-Garcia-Polo, 2017-09-29:
The template files describe the following default Reset_Handler for ARMCC compiler:
1. Call SystemInit() function, that updates at least SystemCoreClock variable. Might update other variables.
2. Call __main(), that among other things copies the initial values or zeroes variables in RAM. Any update done in SystemInit() will therefore be lost.

Issue not present in GCC since data initializing and bss data zeroing is done before calling SystemInit().

JonatanAntoni, 2017-10-12:
Hi @David-Garcia-Polo,

I haven taken your feedback as an opportunity to enhance the documentation accordingly. The recommended way is to configure the system clock during SystemInit() and set SystemCoreClock accordingly. And in main() one should call SystemCoreClockUpdate() to assure SystemCoreClock contains the correct value before any further (dependent) initialization steps are executed.

May I ask you to take a look to the documentation and let us know if its better to understand now, please?

Thanks,
Jonatan

JonatanAntoni, 2018-01-08:
This questions should be answered with the latest documentation. Please feel free to reopen if you have additional questions.

closed 2018-01-08 by JonatanAntoni