ARM-software/CMSIS_5 #251: New event in the I2C driver

koshkin-sergey, 2017-10-02:
I suggest adding a new ARM_I2C_EVENT_SLAVE_BUF_EMPTY event in the I2C driver. It is often necessary to read an arbitrary number of bytes from a slave device. Therefore, the slave must be able to update its transmit buffer during transaction. This will allow us to implement an autoincrement feature like in EEPROM. ARM_I2C_EVENT_SLAVE_BUF_EMPTY occurs when the slave buffer is emptied during the transaction.
[Example.zip](https://github.com/ARM-software/CMSIS_5/files/1349752/Example.zip)

koshkin-sergey, 2017-10-04:
Example of use:

```
void I2C_callback(uint32_t event)
{
  if (event == ARM_I2C_EVENT_SLAVE_TRANSMIT) {
    offset_addr_ptr = 0U;
    prepare_tx_buf(addr_ptr, &tx_buf);
    i2c->SlaveTransmit((uint8_t *)&tx_buf, sizeof(tx_buf));
  }
  else if (event == ARM_I2C_EVENT_SLAVE_BUF_EMPTY) {
    offset_addr_ptr += sizeof(tx_buf);
    prepare_tx_buf(addr_ptr + offset_addr_ptr, &tx_buf);
    i2c->SlaveTransmit((uint8_t *)&tx_buf, sizeof(tx_buf));
  }
  else if (event == ARM_I2C_EVENT_SLAVE_RECEIVE) {
    i2c->SlaveReceive((uint8_t *)&rx_buf, sizeof(rx_buf));
  }
  else if (event & ARM_I2C_EVENT_TRANSFER_DONE) {
    ARM_I2C_STATUS i2c_status = i2c->GetStatus();
    int32_t cnt = i2c->GetDataCount();

    if (i2c_status.direction == 0U) {
      addr_ptr += cnt;
    }
    else {
      addr_ptr = rx_buf.addr;

      if (cnt > 1) {
        write_data(addr_ptr, &rx_buf.data[0], cnt-1);
      }
    }
  }
}
```
Perhaps it is better to define a new event like **ARM_I2C_EVENT_SLAVE_BUF**.
**ARM_I2C_EVENT_SLAVE_BUF** occurs when either the slave buffer is emptied during the transmit transaction or it is full during the receive transaction.

JonatanAntoni, 2017-10-09:
Hi @koshkin-sergey,

Many thanks for your contribution.
Please let me ask some further questions to clearly understand your needs.

According to the current [I2C Driver documentation](http://www.keil.com/pack/doc/CMSIS/Driver/html/group__i2c__interface__gr.html#gafe164f30eba78f066272373b98a62cd4) the event ARM_I2C_EVENT_SLAVE_TRANSMIT is always used whenever the master requests data from a slave. What exactly differentiates the new event you suggest from the existing one?

According to the example you provided you have a slightly different application code behind to realize what you call "auto increment". The current implementations of all the available CMSIS I2C Drivers simply use ARM_I2C_EVENT_SLAVE_TRANSMIT in both cases, i.e. at the beginning of a new transmit and if the master requests more data than already prepared. In order to not break all the available implementations I would like to question whether a solution on application side is possible. Which additional information do you need in order to distinguish between those cases on application level?

For the EEPROM you referenced (as far as I remember) one needs to issue a write request to setup the read position first. All consecutive read requests result in transmitting data with auto increment of the read position. Is that a way you could go for?

Best regards,
Jonatan

koshkin-sergey, 2017-10-13:
Hi @JonatanAntoni,

Thanks for the help.

I rewrote the driver and call-back function code so that only ARM_I2C_EVENT_SLAVE_TRANSMIT event is used. Solution on application side is possible.
```
static
void I2C_callback(uint32_t event)
{
  if (event == ARM_I2C_EVENT_SLAVE_TRANSMIT) {
    if (slave_transfer & SLAVE_TX_TRASFER) {
      offset_addr_ptr += sizeof(tx_buf);
      prepare_tx_buf(addr_ptr + offset_addr_ptr, &tx_buf);
    }
    else {
      offset_addr_ptr = 0U;
      slave_transfer |= SLAVE_TX_TRASFER;
    }

    i2c->SlaveTransmit((uint8_t *)&tx_buf, sizeof(tx_buf));
  }
  else if (event == ARM_I2C_EVENT_SLAVE_RECEIVE) {
    i2c->SlaveReceive((uint8_t *)&rx_buf, sizeof(rx_buf));
  }
  else if (event & ARM_I2C_EVENT_TRANSFER_DONE) {
    ARM_I2C_STATUS i2c_status = i2c->GetStatus();
    int32_t cnt = i2c->GetDataCount();

    slave_transfer = 0U;

    if (i2c_status.direction == 0U) {
      addr_ptr += cnt;
      prepare_tx_buf(addr_ptr, &tx_buf);
    }
    else {
      addr_ptr = rx_buf.addr;
      prepare_tx_buf(addr_ptr, &tx_buf);

      if (cnt > 1) {
        write_data(addr_ptr, &rx_buf.data[0], cnt-1);
      }
    }
  }
}
```
I did not read the documentation very carefully.

closed 2017-10-13 by koshkin-sergey