ARM-software/CMSIS_5 #253: some cosmetic modification in cmsis_armcc.h, cmsis_armclang.h, cmsis_gcc.h, cmsis_iccarm.h

CanastraRF, 2017-10-05:
Dear Support
I found some difference in the cmsis_armcc.h, cmsis_armclang.h, cmsis_gcc.h, cmsis_iccarm.h

1) different parameter
cmsis_armcc.h, cmsis_armclang.h, cmsis_gcc.h
`int16_t __REVSH(int16_t value)
`
cmsis_iccarm.h
`int32_t __REVSH(int32_t val)
`

2) lower, missing U
 cmsis_iccarm.h
```
 __STATIC_INLINE uint8_t __CLZ(uint32_t data) {
    if (data == 0u) { return 32u; }
                 ^             ^ <-- upper U
    uint32_t count = 0;
    uint32_t mask = 0x80000000;
                              ^  <-- missing U
    while ((data & mask) == 0)
    {
      count += 1u;
                ^                <-- upper U
      mask = mask >> 1u;
    }                 ^          <-- upper U
    return (count);
  }

```

3) different comment to   __REV
cmsis_armcc.h, cmsis_armclang.h
`  \details Reverses the byte order in integer value.
`
cmsis_gcc.h  
`  \details Reverses the byte order in unsigned integer value.
`

4) different comment to __REV16
cmsis_armcc.h, cmsis_armclang.h
`  \details Reverses the byte order in two unsigned short values.
`  
cmsis_gcc.h  
`  \details Reverses the byte order in unsigned short value.
`
Please Change this i neer future
Reto

JonatanAntoni, 2017-10-09:
Thanks for making us aware of these issues. I'll update the files soon.

TTornblom, 2017-10-13:
Looking into cmsis_iccarm.h, and comparing it with the definition in https://www.keil.com/pack/doc/CMSIS/Core/html/group__intrinsic__CPU__gr.html#ga1ec006e6d79063363cb0c2a2e0b3adbe, it appears that the definition in cmsis_iccarm.h is correct.

I can fix the "u" vs "U" issues.

TTornblom, 2017-10-17:
Looks like the CMSIS definition of __REVSH() is wrong, according to this page the types should indeed be int16_t. I'll fix this too.

JonatanAntoni, 2017-10-17:
Hi @CanastraRF,

I updated all the header files. Can you please double check if
1. According to ACLE it is "int16_t __REVSH(int16_t value)".
2. Fixed type suffixes to be MISRA compilent.
3. Updated inline and CMSIS documentation.
4. Updated inline and CMSIS documentation.

Thanks,
Jonatan

TTornblom, 2017-10-17:
There seems to be some inconsistent definitions on the __REVSH intrinsic.

According to http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0553a/BABJJDDB.html
the REVSH instruction "Reverse byte order in the bottom halfword, and sign extend to 32 bits", which is also the observable behavior of a Cortex-M4 processor.

The wording "bottom halfword" implies, at least to me, that the argument is a fullword, or int32_t.

So according to that, the definition would actually be:
```c
int32_t __REVSH(int32_t);
```

JonatanAntoni, 2017-10-17:
Hi @TTornblom,

yes, the actual implementation (in hardware) runs on full 32bit register value. It swaps LLSB with LSB and sign-extends the result afterwards. So for instance 0x00000080 becomes 0xFFFF8000. The MMSB and MSB from operand are dropped. So for example 0xFFFF0001 becomes 0x00000100.

According to [ACLE](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ihi0053d/index.html) it is defined as:
`int16_t __REVSH(int16_t);`
which reveals for the above example: 0x0080 becomes 0x8000 and 0x0001 becomes 0x0100. If one assigns an 32bit operand it is truncated to 16bit nonetheless. If one assigns the int16_t result to an int32_t variable sign extension takes place as well.

If there is no other reason I would prefer using ACLE version. This one limits the API upfront to a feasible value range.

Cheers,
Jonatan

CanastraRF, 2017-10-17:
Hi @JonatanAntoni 

> > 2.Fixed type suffixes to be MISRA compilent.

I check the Suffixes.
Mostly are fixed but in cmsis_iccarm.h there is still one line with lower u

```
    __STATIC_INLINE uint8_t __CLZ(uint32_t data) {
      if (data == 0u) { return 32u; }

```

> 1.According to ACLE it is "int16_t __REVSH(int16_t value)".
> 3.Updated inline and CMSIS documentation.
> 4.Updated inline and CMSIS documentation.

Seems to be fine

But now the Brace placement in cmsis_iccarm.h changed to **K&R** styles.
([https://en.wikipedia.org/wiki/Indentation_style])
some inline function has got '**Allman**' style (I prefered)
cmsis_armcc.h
```
__STATIC_INLINE uint32_t __get_CONTROL(void)
{
  register uint32_t __regControl         __ASM("control");
  return(__regControl);
}

```
and other has got **'K&R'** styles.

```
__attribute__((always_inline)) __STATIC_INLINE int32_t __SSAT(int32_t val, uint32_t sat)
{
  if ((sat >= 1U) && (sat <= 32U)) {
    const int32_t max = (int32_t)((1U << (sat - 1U)) - 1U);
    const int32_t min = -1 - max ;
    if (val > max) {
      return max;
    } else if (val < min) {
      return min;
    }
  }
  return val;
}

```
Please use only one style for all inline functions.

JonatanAntoni, 2017-10-17:
Next try ;)

CanastraRF, 2017-10-18:
Thank You very much.
Reto

closed 2017-10-18 by CanastraRF