ARM-software/CMSIS_5 #254: E306: Execution problem

juliobc, 2017-10-06:
Hi,

I have the "E306: Execution problem" with RTX5 and newest MDK. I tried install the patch 

http://www.keil.com/support/docs/3952.htm

but it shows that xerces-c_3_0.dll file is not found. With the last MDK it solved the problem, but now it can not.
Any idea to solve the problem?
Thank you.

juliobc, 2017-10-06:
some info about my MDK:

IDE-Version:
µVision V5.24.2.12
Copyright (C) 2017 ARM Ltd and ARM Germany GmbH. All rights reserved.

Tool Version Numbers:
Toolchain:        MDK-ARM Essential: 1 user(s)  Version: 5.24.1
Toolchain Path:    C:\Keil_v5\ARM\ARMCC\Bin
C Compiler:         Armcc.exe        V5.06 update 5 (build 528)
Assembler:          Armasm.exe        V5.06 update 5 (build 528)
Linker/Locator:     ArmLink.exe        V5.06 update 5 (build 528)
Library Manager:    ArmAr.exe        V5.06 update 5 (build 528)
Hex Converter:      FromElf.exe        V5.06 update 5 (build 528)
CPU DLL:               SARMCM3.DLL          V5.24.1
Dialog DLL:         DCM.DLL              V1.16.0.0
Target DLL:             STLink\ST-LINKIII-KEIL_SWO.dll      V3.0.1.0
Dialog DLL:         TCM.DLL              V1.32.0.0

jkrech, 2017-10-06:
This is a problem of the MDK product and you should contact product support via 
https://developer.arm.com/support
Thanks

~ Joachim

juliobc, 2017-10-06:
ok, thank you

closed 2017-10-06 by juliobc