ARM-software/CMSIS_5 #255: Context Switch - Cortex M23

deepikabhavnani, 2017-10-06:
I have few doubts in SVC context switch implementation. 
1. https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/GCC/irq_armv8mbl.S#L84
Are we trying to store link register in stack? If yes isn;t the sequence incorrect we should assign LR to R7 first and than push R7 to stack.
Same goes for the reverse pop action as well. 
2. Similar issue here as well. https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/GCC/irq_armv8mbl.S#L227

RobertRostohar, 2017-10-09:
> Are we trying to store link register in stack?

No (user LR is automatically pushed/popped by exception entry/exit).

The mentioned line of code saves registers R1,R2,R3 (needed later) before calling TZ_StoreContex_S which destroys R0..R3. A few lines later the registers R1,R2,R3 are restored. R7 is also pushed but just to ensure 8-byte stack alignment.

deepikabhavnani, 2017-10-09:
@RobertRostohar Thanks for clarification. 
LR is automatically popped, but I see after TZ_StoreContex_S we update LR with R7 before we actually pop, will it contain any logical data? R7 is not destroyed?
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/GCC/irq_armv8mbl.S#L87

deepikabhavnani, 2017-10-10:
Thanks for clarification.

closed 2017-10-10 by deepikabhavnani