ARM-software/CMSIS_5 #256: Trustzone RTOS example fails for Cortex-M23

deepikabhavnani, 2017-10-09:
Steps to reproduce:
1. Change the device for secure/non-secure project to ARMCM23_TZ
2. Compile and execute

Switching between different threads is not working and causes fault.

135: ; Dummy Exception Handlers (infinite loops which can be modified) 
   136:  
   137: NMI_Handler     PROC 
   138:                 EXPORT  NMI_Handler               [WEAK] 
0x0000016A 4700      BX       r0
0x0000016C E7FE      B        __aeabi_memcpy4 (0x0000016C)
0x0000016E E7FE      B        0x0000016E
0x00000170 E7FE      B        0x00000170
0x00000172 E7FE      B        0x00000172
0x00000174 E7FE      B        0x00000174

RobertRostohar, 2017-10-10:
The Trustzone RTOS example is configured for Cortex-M33. Changing it to run on Cortex-M23 requires a few steps (it is not enough just to change the device).

Required steps to update the example for Cortex-M23:
-	Set secure project as active and update it:
o	Select new device ARMCM23_TZ
o	Configure debug for Cortex-M23
-	Set non-secure project as active and update it:
o	Select new device ARMCM23_TZ
o	Configure debug for Cortex-M23
o	Configure target memory to use IROM2 and IRAM2 (by default IROM1/IRAM1 is selected)
-	Execute batch build

After that the example runs as expected.

deepikabhavnani, 2017-10-10:
I tried as per above steps, still getting hardfault.
My project settings for secure / non-secure project are:
1. Device -> ARMCM23_TZ
2. Target -> IROM1 / IRAM1 for secure and IROM2 / IRAM2 for non-secure
3. Debug -> -pCM23
Models ARMv8-M Debugger 
C:\Keil_v5\ARM\FVP\MPS2_Cortex-M\FVP_MPS2_Cortex-M23_MDK.exe
C:\Keil_v5\ARM\FVP\MPS2_Cortex-M\ARMCM23_TZ_config.txt

closed 2017-10-10 by deepikabhavnani

RobertRostohar, 2017-10-11:
It should work. Can you provide your complete project?

deepikabhavnani, 2017-10-11:
It worked, I had some local changes in assembly files which caused this mess.