ARM-software/CMSIS_5 #258: arm_mat_inverse_f* functions should document that source gets modified

girtsf, 2017-10-13:
The documentation for matrix inversion functions (http://arm-software.github.io/CMSIS_5/DSP/html/group__MatrixInv.html) does not mention that the source argument will be modified.

There's even a sneaky `const` in `arm_mat_inverse_f64 (const arm_matrix_instance_f64 *pSrc, arm_matrix_instance_f64 *pDst)` that implies that `pSrc` won't be modified. In practice, `const` doesn't help here, since `pSrc->pData` gets modified, not `pSrc` members.

There's a [comment](https://github.com/ARM-software/CMSIS_5/blob/79930bbd0829d816ac55037cb5263c2792fb0f87/CMSIS/DSP/DSP_Lib_TestSuite/Common/src/matrix_tests/mat_inverse_tests.c#L23) in the matching test file that agrees as well:
```
            /* arm_mat_inverse_f32() modifies its source input. Use the scratch
             * buffer to store a copy of the intended input. */
```

It might be helpful to mention that `pSrc` will be clobbered.

llefaucheur, 2017-11-14:
Thank you for pointing to this issue. We will remove the "const" qualifier and add a comment.
By the way we also miss the arm_mat_init_f64(..)

christophe0606, 2020-01-22:
Corrected in commit https://github.com/ARM-software/CMSIS_5/commit/2b1d128349e84c1e5e9803d5d337cf3c338b8d0f

closed 2020-01-22 by christophe0606