ARM-software/CMSIS_5 #261: Wrong code generated for Cortex-M0/M0+

eldarkg, 2017-10-13:
See bug related with CMSIS: [link](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1722849)

JonatanAntoni, 2017-10-16:
Hi @eldarkg,

Thanks for letting us know.

I added "memory" clobbers to the function.
Can you verify that this change solves the issue?

Cheers,
Jonatan

eldarkg, 2017-10-16:
Hi @JonatanAntoni. I post link to this thread to LP. (As I post link from LP to here, threfore I cant test this)

quantum-leaps, 2017-10-17:
I have tested the new "cmsis_gcc.h" (commit 3520562) and it seems to fix the problem. See also post #16 in:
[https://bugs.launchpad.net/gcc-arm-embedded/+bug/1722849](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1722849)
--MMS

JonatanAntoni, 2017-10-17:
Thanks for double checking.

closed 2017-10-17 by JonatanAntoni

kjbracey-arm, 2019-01-08:
Not really planning any action, but adding this note for the record.

@JonatanAntoni I've been checking through "volatiles" and memory clobbers on the assembler, and noticed this one as an apparently unnecessary clobber. It seems that it could now be made conditional on GCC version, as a fix was applied to GCC here https://gcc.gnu.org/bugzilla/show_bug.cgi?id=82602.

Version check would be awkward - looking at dates, I think that would have gone to GCC 6.5, 7.2 and 8.1.

kjbracey-arm, 2019-01-08:
Actually, on reflection, leaving the memory clobber but removing the `volatile` might actually be preferable. We don't actually need to do the `MRS` read if we don't need the result.

You need only 1 of `volatile` and `memory`.

| |We require|volatile|memory clobber|
|---|---|---|---|
|Ordered vs CPSID/CPSIE|yes|yes|yes|
|Ordered vs normal memory accesses|no|no|yes|
|Instruction must be issued even if result not used|no|yes|no|

So we can take one of the qualifications off, but it's not obvious to me which is better.

Taking off the volatile would help people who use "disable IRQ + return old state" calls but don't bother to check the return value - eliminates the MRS.

Taking off the memory clobber maybe helps scheduling a little, but unlikely to make much difference, assuming that it's probably accompanied by a memory-clobbering CPSID most of the time anyway.