ARM-software/CMSIS_5 #266: An 8-bit reserved area with an offset of 0x2 in the 32-bit aligned address on SVDConv

tu44okawa3, 2017-10-23:
# Problem
When I compiled an 8-bit reserved area with an offset of 0x2 with SVDConv.exe, RESERVED will be output with uint16_t. However, there is no problem when the offset is 0x0, 0x1, or 0x3.

## Procedure to reproduce the problem
Compile the SVD containing the following code:

        
        <register>
          <dim>2</dim>
          <dimIncrement>1</dimIncrement>
          <dimIndex>0,1</dimIndex>
          <name>DMMY4%s_8B</name>
          <description>8bit register</description>
          <addressOffset>0x6C</addressOffset>
          <size>8</size>
          <access>read-write</access>
        </register>
        <register>
          <name>DMMY43_8B</name>
          <description>8bit register</description>
          <addressOffset>0x6F</addressOffset>
          <size>8</size>
          <access>read-write</access>
        </register>

                
        <register>
          <name>DMMY5_16B</name>
          <description>16bit register</description>
          <addressOffset>0x70</addressOffset>
          <size>16</size>
          <access>read-write</access>
        </register>
        <register>
          <name>DMMY53_8B</name>
          <description>8bit register</description>
          <addressOffset>0x73</addressOffset>
          <size>8</size>
          <access>read-write</access>
        </register>

## Error results
    __IOM uint8_t   DMMY40_8B;                    /*!< (@ 0x0000006C) 8bit register                                              */
    __IOM uint8_t   DMMY41_8B;                    /*!< (@ 0x0000006D) 8bit register                                              */
    __IM  uint16_t  RESERVED8;
    __IOM uint8_t   DMMY43_8B;                    /*!< (@ 0x0000006F) 8bit register                                              */
    __IOM uint16_t  DMMY5_16B;                    /*!< (@ 0x00000070) 16bit register                                             */
    __IM  uint16_t  RESERVED9;
    __IOM uint8_t   DMMY53_8B;                    /*!< (@ 0x00000073) 8bit register                                              */

## Expected results
    __IOM uint8_t   DMMY40_8B;                    /*!< (@ 0x0000006C) 8bit register                                              */
    __IOM uint8_t   DMMY41_8B;                    /*!< (@ 0x0000006D) 8bit register                                              */
    __IM  uint8_t   RESERVED8;
    __IOM uint8_t   DMMY43_8B;                    /*!< (@ 0x0000006F) 8bit register                                              */
    __IOM uint16_t  DMMY5_16B;                    /*!< (@ 0x00000070) 16bit register                                             */
    __IM  uint8_t   RESERVED9;
    __IOM uint8_t   DMMY53_8B;                    /*!< (@ 0x00000073) 8bit register                                              */

# Reproducible svd file
I used the modified attached ARM_Example_8bit.svd

# Environment info
- OS: Win10
- SVDConv version:  V3.3.9
- Execute command :  <code>SVDconv.exe ARM_Example_8bit.svd --generate=header --debug-headerfile</code>
[ARM_Example_8bit.svd.txt](https://github.com/ARM-software/CMSIS_5/files/1406217/ARM_Example_8bit.svd.txt)

JonatanAntoni, 2017-10-23:
Hi @tu44okawa3,

Thanks for making us aware of this issue.
May I ask you to give the attached version 3.3.11 a try, please?

Best,
Jonatan

[SVDConv-3.3.11.zip](https://github.com/ARM-software/CMSIS_5/files/1406827/SVDConv-3.3.11.zip)

tu44okawa3, 2017-10-23:
Hi Jonatan,

Thanks to you, I was able to get the results I expected.
Version 3.3.11 resolved the problem.

Many thanks,
tu44okawa3

closed 2017-10-23 by tu44okawa3