ARM-software/CMSIS_5 #268: uxTaskGetSystemState() FreeRTOS equivalent in RTX5?

strongleap, 2017-10-26:
Hi, 

I need to get list of all running/available tasks from a task that does check the integrity of the system.

I need to check status of each task then.

Has RTX5 got any API to request the list of all tasks in the system?

I am looking for something like uxTaskGetSystemState() equivalent in Free RTOS.

Thanks,
Anil

JonatanAntoni, 2017-10-26:
Hi Anil,

Yes, you can get a list of threads with `osThreadEnumerate`, refer to [Thread Management](http://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html).

Enumerate populates a given array with all known thread ids. Please note that the given array needs to provide enough slots otherwise you receive only a subset. You can use `osThreadGetCount` to query the amount of running threads in advance.

Cheers,
Jonatan

JonatanAntoni, 2018-01-08:
I think this question is answered. Please feel free to reopen if you have additional questions.

closed 2018-01-08 by JonatanAntoni