ARM-software/CMSIS_5 #269: Unexpected timer performance with different thread configuration

NrcsRob, 2017-10-27:
Hello,
We are facing the issue with the osTimer when running with the different osThread configuration.
Here are the configurations:
Configuration1:
created 6 osThread and a 2ms osTimer
with OS_TICK_FREQ = 1000, OS_ROBIN_TIMEOUT = 3, TimerThread Priority as RealTime
=>Result: **This is OK**
https://drive.google.com/open?id=0B4BgoSgZmc47Q2dCWnpMSFFKakk
(yellow, green, purple line is osThread switching IO, cyan is the 2ms IO toggling)

Configuration2:
**created 5 osThread** and a 2ms osTimer
with OS_TICK_FREQ = 1000, OS_ROBIN_TIMEOUT = 3, TimerThread Priority as RealTime
=>Result: **NOT OK** 
https://drive.google.com/open?id=0B4BgoSgZmc47T1F4T3gtYkZCZWM
(cyan is the 2ms IO toggling with unexpected toggling)

inside the threads and osTimer, it is just doing a single I/O toggling.
It seems if we select RTX5 source instead of library that suggested by the workaround in CMSIS 5.1.0, then it works, but somehow the problems appear again when running with 3 thread (but 3 thread is worked with rtx5 source selected).
We are using released version CMSIS 5.1.1

JonatanAntoni, 2017-10-27:
Hi @NrcsRob,

Its hard tell what might be the reason here.

If the timer thread has the highest priority in the system, the timer callbacks should be accurate no matter what other threads are doing. Please note that mixing priority scheduling and round robin has a known issue #228. 

Is it feasible for you to create an event log using EventRecorder? Or would you be able to provide a reproducer project we can take a look into?

Best regards,
Jonatan

NrcsRob, 2017-11-03:
Hi Jonatan,

You may play around with the task definition to turn on/off the task
#define TASK3 1
#define TASK4 1
#define TASK5 1
#define TASK6 1

In most cases, the osTimer will perform inconsistently if system running with 3 or 5 thread.
We tried the cases like running it with Library or with Source, Start the timer in a task or start the timer during initialization. The osTimer will behave differently with those configuration, while we are expecting it to be a consistent 2ms timing event.

[RTOSVer2_SystemTest.zip](https://github.com/ARM-software/CMSIS_5/files/1440455/RTOSVer2_SystemTest.zip)

Thank you for your support

JonatanAntoni, 2017-11-06:
Hi @NrcsRob,

thanks for the reproducer. We will take a look into.

Best,
Jonatan

JonatanAntoni, 2017-11-13:
Hi @NrcsRob,

I have taken a quick look into your example.
It looks like you have multiple issues. But even though I cannot reproduce the issues you see. I tried running your example on a STM32F746G-Discovery. 

Please try to clean up your startup and initialization code to fulfill the requirements of RTX5. 
- HAL and Clock initialization should be done before osKernelInitialize. RTX5 assumes clock and interrupt settings not to change while kernel is running.
- Assure to remove all STM HAL library initialization/usage to RTOS controlled resources, like SysTick, PendSV and SVC.

Next you could try to strip down the example to a bare minimum, i.e. remove manual thread attribute settings if not absolutely needed, i.e. all the threads should be operational if started with default attributes. You might need to configure the default thread stack size appropriately. Also try to activate thread stack watermarking to check if thread stacks are really large enough.

Furthermore you might want to try setting explicit thread priorities other than normal, i.e. assign each thread to its own priority level and avoid round robin scheduling. Instead of letting all threads run as fast as possible try to insert some blocking statements, e.g. osDelay.

Please let me know if one of the above helps. I think its much easier to enhance the application from a known point instead of digging in the dark. Consider posting to the STM32 community as well because it might be an integration issue with the HAL library you are using.

Best,
Jonatan

JonatanAntoni, 2018-01-08:
Hi @NrcsRob,

Did any of my above hints help? Or do you still have concerns you'd like to discuss?

Best,
Jonatan

JonatanAntoni, 2018-02-06:
Closing this issue. Please feel free to reopen if further discussion is needed.

closed 2018-02-06 by JonatanAntoni