ARM-software/CMSIS_5 #274: CMSIS SPI Driver

joabeck, 2017-11-03:
Hello,
i am trying to implement the CMSIS SPI driver for a new processor platform.
According to the SPI driver description (picture under Reference/SPIDriver showing a master with 3 slaves), it should be possible to use the driver with different select signals (SS).
Here I miss a function in the driver which selects the SS signal for the next Transfer call.
Does this mean, I have to implement a function which goes around the CMSIS SPI driver which selects the SS signal in the hardware driver?
If this is right, I thing there is something missing in the driver specification (for example a new control function like ARM_SPI_SELECT_SS).
Thank you for your support.
Best regards
Joachim

RobertRostohar, 2017-11-06:
SPI peripherals have typically hardware support for a single Slave Select (SS) line. Therefore also the CMSIS SPI driver provides support for a single SS line. Multiple SS lines support was considered but turned out to only complicate the SPI driver specification and especially implementation.

Suggested alternative is to export a driver instance per SS line which would handle the SS line and call the common driver (also taking care of concurrent access by using semaphores for example).

Template which shows such example is planned to be added to the documentation.

JonatanAntoni, 2019-03-12:
Closing due to inactivity. @joabeck Please let us know if you want to discuss further aspects.

closed 2019-03-12 by JonatanAntoni

herrmanthegerman, 2022-03-03:
Of course I'd like to see the CMSIS driver to support multiple slave select lines (e.g., slave is selected with each call of a data transfer command). 

In absence of this feature, I think the driver documentation should mention explicitly that it does not support multiple slave select lines. I'd suggest also to proivde an example how to work around it. 

Thanks.

ecorini, 2022-04-15:
I also think that a control like `ARM_SPI_SELECT_SS` should be supported, of curse it will be HW dependent but if only one chip select is present, the driver could simply return `ARM_DRIVER_ERROR_UNSUPPORTED` .

Thanks