ARM-software/CMSIS_5 #276: Newest version of osDelayUntil is subject to a race condition

graeme-hattan, 2017-11-06:
Hello,

We have had some problems with the newest version (V2.1.1) of the osDelayUntil function. Our analysis indicates the new behaviour of the function introduces a race condition which makes it unsafe to use for periodic timing, please see the description below for more information.

The previous version of the documentation (V2.1.0) for the osDelayUntil function states the following for the return value:

> osParameter: the time is either in the past or cannot be handled (out of bounds).

However this is not the case for the current version (V2.1.1). Here if the function is called with a value of more that 1 tick in the past (looking at the implementation, I see that it will still return osErrorParameter if it is exactly 1 tick in the past), it simply delays until the kernel ticks wrap to that value.

This is a problem since there is no way to guarantee that the tick value has not passed by the time the function enters its svc call.

Considering the example code given for the function:

```
#include "cmsis_os2.h"
 
void Thread_1 (void *arg)  {                // Thread function
  uint64_t tick;
  tick = osKernelGetTickCount();            // retrieve the number of system ticks
  for (;;) {
    tick += 1000;                           // delay 1000 ticks periodically
    osDelayUntil(tick);
    // ...
  }
}
```

If `Thread_1` is pre-empted by a higher priority task and ends up in waiting the READY state for more that 1001 ticks, the value of the `tick` variable will fall behind the kernel tick count. This will cause to osDelayUntil delay for a duration that is around 50 days for a 1 ms tick.

The likelihood of the issue can be reduced by incrementing `tick` until it is in the future, eg:

```
#include "cmsis_os2.h"
 
void Thread_1 (void *arg)  {                // Thread function
  uint64_t tick;
  tick = osKernelGetTickCount();            // retrieve the number of system ticks
  for (;;) {
    while(tick < osKernelGetTickCount())
      tick += 1000;                           // delay 1000 ticks periodically
    // Pre-emption still possible here or in first stages of osDelayUntil
    osDelayUntil(tick);
    // ...
  }
}
```

However, as shown, `Thread_1` can still be pre-empted before osDelayUntil enters its svc call.

For us, this has been problematic since have been using small increments in the tick value. With the previous behaviour of osDelayUntil there was no problem as osParameter would be returned, however now the function is unsafe to use because of this race condition. Even if our thread is the highest priority thread in the system, there is still be possibility of the time being taken up by chained interrupts.

I would suggest that osDelayUntil should return osParameter if the delay calculated is greater than half the maximum number of ticks (ie 0x7FFFFFFF). This would make it safe to use for periodic delays as long as the interval is less than around 25 days on a system with 1 ms ticks.

Thank you for taking the time to consider this issue.

Graeme

kjbracey-arm, 2017-11-07:
I see another potential failure mode apart from higher-priority threads and chained interrupts - the kernel being suspended.

If someone puts the system to sleep manually using `osKernelSuspend()` for longer than it would like, threads are going to have to play catch-up afterwards. The RTX kernel clearly copes with this sort of manual suspend, and threads can, but anyone using `osDelayUntil` like in its documentation example would fall straight over. Rather than going round the loop without delay for each 1000 ticks of suspend until it reaches current time, it would immediately block for 2^32.

kjbracey-arm, 2017-11-07:
Aside from the late issue, I do have to ask - why is a "0.0.1" version increment making such a big API change with the time type? What was the aim? This is a headache in any codebase which has to make a 2.1.0->2.1.1 transition, especially as there's no easy way to detect the API change - there's no API version number macro in cmsis_os2.h.

I always kind of understood looking at the design that CMSIS-RTOS was supposed to allow binary compatibility - apps compiled against cmsis_os2.h could work with any implementation if they used dynamic allocation. But now 2.1.0 and 2.1.1 have this binary incompatibility. At least for this API, it may as well be CMSIS-RTOS 3.

What was wrong with the 64-bit timebase? There was an implementation limit for osDelayUntil in that it couldn't wait more than 2^32, but that could have easily been lifted by making multiple internal wait calls. But I see no problem with the actual public 2.1.0 64-bit API.

I can see discussion about widening it to 64-bit for CMSIS-RTOS 2 - https://github.com/ARM-software/CMSIS/issues/44 but see no discussion about narrowing it again.

RobertRostohar, 2018-05-21:
Longest possible delay for osDelayUntil is now limited as suggested.

graeme-hattan, 2018-06-26:
Looks good to me, thanks.

closed 2018-06-26 by graeme-hattan