ARM-software/CMSIS_5 #277: osDelayUntil - problem if we are late

kjbracey-arm, 2017-11-06:
I had been working with the 2.1.0 API, assuming kernel tick count was 64-bit.

I see now that 2.1.1 has changed that to 32-bit. That's not a problem in itself (aside from the general headache of API instability).

But there is a problem in the detail of osDelayUntil.

In the 64-bit version, if you manage to overrun your target time, there's no problem using osDelayUntil:

* we were scheduled to wake at 0x8000 - it's 0x8000 now, or maybe we're a bit late
* compute target next wake time = 0x8000 + 10 = 0x800a
* do some work (or get pre-empted by higher-priority thread), so that we're now definitely late
* time now is 0x800c
* osDelayUntil(0x800a)
* Call returns immediately with "parameter" error - we're behind schedule but running

The same sequence with the 32-bit version will have osDelayUntil decide the thread is asking to be delayed for 2^32-2 ticks and put it to sleep, rather than making the more natural assumption that we've overrun by 2 ticks and are late.

The 64-bit API regards the number space as "1 value present, 2^32-2 values future, remaining 2^64-2^32 values invalid as either past or too far in future".

The 32-bit API regards the number space as "1 value present, 2^32-2 values future, 1 value invalid as too far in future".

For an "until" API to be safe and useful, it really needs to cope with times in the past being requested - this can happen. osDelayUntil already supports 1 specific already-past value, ie the current tick count, for which it returns osOK. Something further in the past than the most recent tick should still return osOK - it's not logical to treat "request for last-passed tick" differently from "request for 3 ticks ago". They're both in the past, it's just a matter of degree.

Otherwise there is no safe way to use the API - there's no solid way to be absolutely certain that the intended wake-up time hasn't already passed by the time you call osDelayUntil, beyond making sure to always request `osKernelGetTickCount() + [more_ticks_than_i_can_possibly_be_preempted_for]`.

You might just want the following to work:

     uint32_t t = osKernelGetTickCount();
     // Loop runs every 10 ticks, without ever drifting
     for (;;) {
           do_work;
           t += 10;
           osDelayUntil(t); 
     }

That's in principle more solid than just using `osDelay(10)`, which can drift and definitely would if work+preemption could take longer than 1 tick, but it is in danger of failure with the the 2.1.1 osDelayUntil(), if work+pre-emption managed to approach 10 ticks.

The problem maybe isn't so extreme for just plain delay, but if you consider other potential "until" APIs where there's another wake-up source, such as a condition variable wait_until or any sort of semaphore with some spurious wakes, it would be absolutely critical that this didn't fail when late:

      uint32_t end_time = osKernelGetTickCount() + timeout;
      while (!condition) {
          if (condvar.wait_until(end_time) == osErrorTimeout) {
              break;
          }
      }

It's very possible that you could end up with a wait_until call made after end_time after a few iterations if other waiters held the mutex too long, and that must return "timeout", not keep going.

I would suggest following the guideline's of RFC 1982 serial number arithmetic - assume "half the values are old and half the values are new".

Something like:

    osStatus_t svcRtxDelayUntil (uint32_t ticks) {

        ticks -= osRtxInfo.kernel.tick;

        if (ticks == 0U || ticks > 0x80000000U) {
            // ticks appears to be in the past
            return osOK;
        }

        osRtxThreadWaitEnter(osRtxThreadWaitingDelay, ticks);

        return osOK;
    }

This does mean the longest possible wait_until is down to 2^31, but you've got to regard some of the 32-bit timestamp space as the past, and the RFC 1982 50/50 split seems like the simplest general rule for timestamp comparisons.

graeme-hattan, 2017-11-06:
This is the same as https://github.com/ARM-software/CMSIS_5/issues/276.

I suggest returning `osErrorParameter` if the time is in the past though, so that the caller knows that it missed the time window.

kjbracey-arm, 2017-11-06:
Looks like we had a race condition on the issue :)

So you've actually hit the problem - I've only just noted it from inspection.

Sounds like we're basically in near-total agreement.

Not sure about the usefulness of the return, in particular anything that distinguishes `osDelayUntil(current_tick_value)`, given that that's already in the past. But that's a minor detail. As long as it returns immediately.

graeme-hattan, 2017-11-06:
A race condition reporting a race condition :)

The return value is useful for error reporting, it is helpful to know if execution is taking longer than expected. Also, if you need to keep in sync with the time increment, it may be preferable to skip an iteration of the loop, and just increment the time and try again. Rather than risk drifting off further.

I suppose this is easy enough to check after return, but why repeat the math if `osDelayUntil` just did it?

kjbracey-arm, 2017-11-07:
Yes, seems a reasonable thing to report - just as long as osDelayUntil(current_tick_value) also returns that error - in that case you already are later than you expected, albeit by less than a tick. I think "late" has to be defined without an arbitrary 1-tick threshold.

graeme-hattan, 2017-11-07:
Possibly. I was thinking a call to `osKernelGetSysTimerCount` would be fine if that level of granularity was needed. OS tick functions generally have an error up to one tick, this is to be expected. Either way would be fine for us though.

graeme-hattan, 2017-11-07:
Er, actually calling `osKernelGetSysTimerCount` would be subject to another race condition if a tick occurs after (or just before) `osDelayUntil` returns.

Any `osKernelGetSysTimerCount` call would have to be accompanied by `osKernelGetTickCount`, and the combination would have to be atomic. I'm not sure this is even possible due to the nature of `osKernelGetSysTimerCount`.

kjbracey-arm, 2017-11-07:
I think you're off on a slight tangent - I'm not talking about levels of granularity, just the fact that if osKernelGetTickCount() says 2000, then the current time must be 2000-and-a-bit. We don't need to know what that bit is, but it's there.

A request for osDelayUntil(2000) is really a request for 2000.00000, and that's when it would wake you, if there was infinite CPU speed. Regardless of phase or granularity, osDelayUntil is supposed to be waking you up at the start of a tick - you should be getting a regular period, with better-than-tick precision - CPU load and thread priorities permitting.

With 1-second ticks, you should be able to use osDelayUntil with 1-tick increments to increment a digital clock face smoothly. If you find yourself doing osDelayUntil(12:15:45) when it's really 12:15:45.50, you've already missed the time you should have updated the clock face, and you're already visibly stuttering. Despite the fact you're asking to delay until the "current" time at the second tick precision.

Now, not being late on entry to osDelayUntil doesn't prove anything - you could still be late by the time you've woken up due to scheduling issues. But if doing an entry lateness test, asking for current time value is definitely late. Asking for a future time value is definitely not late, at least at the instant it's checked.

Viewing asking for current time value as not late is an off-by-one error, which can't be excused on the basis of one-tick granularity. osDelay's time can be excused being off by up to one due to tick phase. osDelayUntil doesn't suffer that problem, so a 1-tick error in its lateness test can't be excused.

kjbracey-arm, 2017-11-07:
BTW, if you do want super-precision tick time, you would do something like

    do {
        tick1 = osKernelGetTickCount();
        subtick = osKernelGetSysTickCount();
        tick2 = osKernelGetTickCount();
     } while (tick1 != tick2);
     combine tick1 and subtick

That works from a thread context or any lower-level interrupt context than the timer tick, such that the systick wrap increments the tick count. See

See first answer to https://stackoverflow.com/questions/5162673/how-to-read-two-32bit-counters-as-a-64bit-integer-without-race-condition

graeme-hattan, 2017-11-07:
Yes, that makes sense. It helps to report any entry lateness.

RobertRostohar, 2018-05-21:
Longest possible delay for osDelayUntil is now limited as suggested.

graeme-hattan, 2018-05-21:
Thanks for putting in the fix, however after some discussion we were in agreement that it should also be an error if the delay time is zero. Eg

	if (0U == ticks || ticks > 0x7FFFFFFFU) {
	    ...
	    return osErrorParameter;
	}

Relevant comments:
https://github.com/ARM-software/CMSIS_5/issues/277#issuecomment-342406287
https://github.com/ARM-software/CMSIS_5/issues/277#issuecomment-342488448

RobertRostohar, 2018-05-22:
osDelayUntil has been updated as suggested.

ReinhardKeil, 2018-05-23:
feature is not implemented. Closing this issue as it is a duplicate with #276.  Please provide further comments under #276

closed 2018-05-23 by ReinhardKeil