ARM-software/CMSIS_5 #278: osMailPut call from ISR

alexander-mu, 2017-11-13:
osMailPut can be called from ISR:

[https://www.keil.com/pack/doc/CMSIS/RTOS/html/functionOverview.html#CMSIS_RTOS_ISR_Calls](https://www.keil.com/pack/doc/CMSIS/RTOS/html/functionOverview.html#CMSIS_RTOS_ISR_Calls)

There should be added a note "Interrupt Service Routines can call this function" to the documentation of this function.

JonatanAntoni, 2017-11-13:
Hi Alexander,

I'll add this hint to the documentation.

Please take not that CMSIS RTOSv1 has been superseded by RTOSv2 and is no longer under active development. For new projects its highly recommended to use CMSIS RTOSv2 as implemented by RTX5.

Thanks,
Jonatan

JonatanAntoni, 2018-01-08:
This issue should be solved thus closing. Please feel free to reopen this issue if further discussion is needed.

closed 2018-01-08 by JonatanAntoni