ARM-software/CMSIS_5 #280: core_cm7.h : Suggestions to improve SCB_CleanDCache_by_Addr and friends

udoe, 2017-11-15:
SCB_CleanDCache_by_Addr, and similar cache manipulation functions, are not easy to use. There are a couple of issues:
- The function requires a cache-line-aligned address. It's easy to make the mistake of not rounding up dsize when rounding down addr.
- The addr argument is typed uint32_t* so a cast is required when an arbitrary pointer type is passed.
- The dsize argument is a signed integer type. Typically it will be called with an unsigned int. So a cast is required.
- If the function is called with a dsize of zero, it does not manipulate cache lines but still performs the side effects of the memory barrier operations.

Below is a proposal for an improved implementation. To maintain backward compatibility, it could use a different name such as SCB_CleanDCache_Region, or could replace SCB_CleanDCache_by_Addr. I don't know your policy regarding API compatibility...

Udo

/**
  \brief   Clean a D-Cache region
  \details Cleans D-Cache for the given address and size.
  \param[in]   addr    address
  \param[in]   dsize   size of memory block (in number of bytes)
*/
__STATIC_INLINE void SCB_CleanDCache_Region(void *addr, uint32_t dsize)
{
  #if defined (__DCACHE_PRESENT) && (__DCACHE_PRESENT == 1U)
    if ( dsize > 0 ) {
      const uint32_t linesize = 32; /* in Cortex-M7 size of cache line is fixed to 8 words (32 bytes) */
      uint32_t op_addr = ((uint32_t)addr) & ~(linesize - 1);
      int32_t  op_size = (int32_t)(dsize + (((uint32_t)addr) & (linesize - 1)));
    
      __DSB();

      do {
        SCB->DCCMVAC = op_addr;
        op_addr +=          linesize;
        op_size -= (int32_t)linesize;
      } while ( op_size > 0 );

      __DSB();
      __ISB();
    }
  #endif
}

JonatanAntoni, 2017-11-15:
Hi @udoe,

thanks for sharing your feedback. I think such a change could be integrated without major compatibility issues. I'll check that and discuss with the dev team.

Best,
Jonatan

alfred-ai, 2017-12-22:
It will be even better if there is a check for the alignment of the starting address.

These cache maintenance API may end up receiving an address from a few layers up, which means a simple check (by `assert`) of `(uint32_t)addr % DEFAULT_CACHELINE` will help catching aligned related problems.

udoe, 2017-12-22:
My main point (item 1 above) was that it is easy to use the function in a wrong way: If you round down the address to meet the alignment requirement it's easy to forget rounding up size properly, or to implement this rounding in the wrong way.

Maybe there should be two layers of functions: SCB_CleanDCache_Lines which requires an aligned address and size, and on top of that SCB_CleanDCache_Region which takes an arbitrary address and size and handles the rounding properly.

GuentherMartin, 2018-10-12:
Hello @udoe please check the modified functions. 
The changes are not exactly your proposal but have the same convenience for the user.
The modified functions do the same as the old functions (if the correct parameters are passed) therefore we used the original function names.
Please close the issue when the changes fulfill you expectation.

udoe, 2018-10-15:
The new function implementations look good too me. Thanks for your efforts.

closed 2018-10-15 by udoe