ARM-software/CMSIS_5 #281: osThreadGetId with IRQs masked

kjbracey-arm, 2017-11-15:
I'm struggling a bit to implement some tricky ISR<->thread synchronisation, and part of the issue is the lack of definition of interactions between CMSIS-RTOS and interrupt disabling.

I see some related discussion here: https://github.com/ARM-software/CMSIS/issues/38

Nothing in the CMSIS-RTOS documentation says anything about whether API calls can be made from thread context with interrupts disabled.

As it's not mentioned, I would assume it's permitted, but no...

In RTX 5, the ARM-M implementation bars such calls with `IS_IRQ_MASKED`, except for those defined as working from interrupt, which it treats as if called from interrupt, sometimes with PendSV firing work as interrupts get re-enabled; the ARM-A implementation permits any call, as `IS_IRQ_MASKED` is defined is 0.

The ARM-M clearly has a number of issues with making SVC calls with IRQs masked - it may be there's no easy solution there.

Really, the documentation needs a bulk update to add "or with interrupts disabled" to every "cannot be called from Interrupt Service Routines", or similar.

But I would like to request a specific exception - allow `osThreadGetId()` to work with interrupts disabled. At the minute, it uses an SVC call with the same `IS_IRQ_MODE() || IS_IRQ_MASKED()` check as all other calls.

But there's no need for any "read current thread ID" to use an SVC - there's no synchronisation issue, as the ID is constant for a given thread - `osRtxInfo.thread.run.curr` changes on context switch, but it always has the same value during execution of any given thread. Therefore it is inherently thread-safe to read. Therefore we can avoid the SVC and make the call work with interrupts disabled. (And be more efficient).

In this case it could be implemented as

    osThreadId_t osThreadGetId (void) {
      osThreadId_t thread;
      if (IS_IRQ_MODE()) {
        thread = NULL;
      } else {
        thread = osRtxThreadGetRunning();
      }
      EvrRtxThreadGetId(thread);
      return thread;
    }

Being assured that `osThreadGetId()` permits calls from critical section would avoid me needing to manually peek `osRtxInfo.thread.run.curr`, making my code CMSIS-RTOS generic.

Background: I'm implementing an ISR->thread wake mechanism that needs some way of identifying the current thread while still in a critical section to achieve atomic unlock-and-wait. With this change, I can add my thread ID to a wait queue before unlocking and issuing osThreadFlagsWait. The ISR notified can take IDs off the queue and call osThreadFlagsSet. It potentially opens the door to other solutions, using the thread ID as some sort of key to thread-local-storage.

kjbracey-arm, 2017-11-15:
It's worth noting that this functionality is only relevant if the OS is in privileged mode so threads can disable interrupts in the first place.

If the SVC would be required for any sort of memory protection purpose on an unprivileged build, then the IRQ masking limitation would be doing no harm there, so you could retain the SVC to access osRtxInfo in that case if really necessary.

Although for efficiency, it would be more conventional anyway to ensure that something like "current thread ID" or "thread-local-storage pointer" is available with a simple unprivileged read, rather than needing a system call.

(How I'd love generic thread-local storage - would be a better overall solution to this problem, but this suggestion is a simple out for me).

ReinhardKeil, 2018-06-21:
Now implemented with https://github.com/ARM-software/CMSIS_5/commit/f7d6092e0947ab71cd53399707aa385db1d01e3a

JonatanAntoni, 2018-06-21:
@kjbracey-arm, does this enhancement solve your request?

kjbracey-arm, 2019-01-17:
Yes, thanks!

closed 2019-01-17 by kjbracey-arm