ARM-software/CMSIS_5 #284: Problem with RTX5 when changing clock (LPC54608)

joabeck, 2017-11-23:
Dears Sirs,

we are using the LPC54608 Expresso54608 eval board (OM13092).

I have created a simple Blinky-Demo with RTX5. Now i see the following problem which i do not understand:

Configuration 1:

- init the main clock to 12MHz in system_LPC54608.c 
- change the clock to 96MHz at the beginning of main
- creating the threads
- osKernelStart
- main goes into endless loop

--> everything is working, the threads are running, the Leds are blinking.

Configuration 2:

- init the main clock to 96MHz in system_LPC54608.c 
- no change of the clock in main
- creating the threads
- osKernelStart
- main goes into endless loop

--> the program runs into the endless loop, but no tasks are executed, the Leds are not blinking.

An analysis shows, that the SysTick_Handler in irq_cm4f.s is not executed in configuration 2. In configuration 1 it starts running after osKernelStart.

The same happens with a clock of 48MHz.

I am using CMSIS 5.2.0

Thank you for your support.

best regards 
Joachim

JonatanAntoni, 2017-11-23:
Hi Joachim,

Do you set SystemCoreClock variable in main, i.e. calling to UpdateSystemCoreClock()?

If you set the core clock during SystemInit (i.e. pre-main) the static variable "SystemCoreClock" is overwritten by c library start code. Thus you need to assure the value presented by SystemCoreClock in main prior to call osKernelStart. osKernelStart uses the value in SystemCoreClock to configure the SysTick appropriately. This might cause undefined behavior if the value in SystemCoreClock is wrong.

Please let me know if my first guess solves your issue. Otherwise I would like to ask you to provide a reproducer project I could take a detailed look into.

Best,
Jonatan

joabeck, 2017-11-24:
Hello Jonatan,
i found the problem. It was on my side. The calculation of SystemCoreClockUpdate delivers under some circumstances a zero result...

Sorry!
Thank you for your support.
Best regards
Joachim

closed 2017-11-24 by joabeck

JonatanAntoni, 2017-11-24:
Hi Joachim,

Thanks for letting us know that it works for you, now.

Its my pleasure.
Best,
Jonatan