ARM-software/CMSIS_5 #285: SVD issue

AmiccomVicChen, 2017-11-24:
Hello~ 
I am a beginner on here, have a trouble need help.
I am creating a Device Family Pack, need to prepare SVD file.
I convert SVD file to header file, always get result1.

how to edit SVD file to get result2 in header file?
Please help me, thanks.

**Result1:**
typedef struct { 
     __IO uint32_t  DATA0;
     __IO uint32_t  DATA1;
     __IO uint32_t  DATA2;
     __IO uint32_t  DATA3;
} GPIO0_Type;

**Result2:**
typedef struct { 
     __IO uint32_t  DATA[4];
} GPIO0_Type;

jkrech, 2017-11-24:
The register description is documented here: http://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html#elem_register
Please take a look at the ARM_Example.svd file located in the CMSIS_5\CMSIS\Utilities\  folder

<register>
  <dim>4</dim> 
  <dimIncrement>4</dimIncrement> 
  <name>DATA[%s]</name> 
  ...
</register>

AmiccomVicChen, 2017-11-24:
Hi Jkrech,
Thank you very much!

closed 2017-12-08 by jkrech