ARM-software/CMSIS_5 #289: System crash issue

jameszxj, 2017-12-02:
I evaluate RTX2 on xilinx z7020 board,  The system crashed when run for a while.
z7020 has dual cortex-A9 core. core0 run RTEMS, core1 run RTX2
test1: core0 send softIRQ to core1 every 2ms, core1 just act the interrupt and flash a led, the system works successfully
test2: core0 send softIRQ to core1, core1 release a semaphore from isr, a task wait at the semaphore and flash a led，after 5-30 minutes, the system wnet down.

JonatanAntoni, 2017-12-04:
Hi @jameszxj,

Thanks for your feedback. I have to admit that we don't have any plans to support RTX (I guess you refer to RTX5) on multi core Cortex-A processors. For sure your scenario sounds feasible and it should work that way.

Can you analyze the issue a bit more detailed? 
- Which version of RTX5 are you using?
- Did you carefully follow the initialization routine for RTX5? I.e. the interrupt priority configuration is crucial and can lead to surprising behavior if done wrong.
- What exactly do you mean with "the system went down"? Are you able to record a trace or EventRecorder output?
- Do you observe any fault exception? Or does RTX5 calls to osRtxErrorNotify? Or does it simply stick in the idle loop?
- Is this effect limited to the use of a Semaphore? What about using a Mutex instead? Or a Message Queue? Or some sort of flags (event or thread flags)?

We need to be more precise here in order to figure out the root cause. Please feel free to provide as many details as you can.

Best,
Jonatan

jameszxj, 2017-12-08:
1.  I use the devlop version from https://github.com/ARM-software/CMSIS_5.git
2.  I tested on freeRTOS, It runs OK. The initializations are the same. I do not notice the information about interrupt in the document, would you please point out it?  Would you mean RTX5 need a config just like FreeRTOS configMAX_API_CALL_INTERRUPT_PRIORITY?
3. The program triggers a led. The led do not flash again.
4. No, I don't. I will do more tests and collect more information about it.
5. I just test it using event, the same things happened.

JonatanAntoni, 2017-12-08:
Hi @jameszxj,

1. Should be fine, even if it might happen that we introduce instabilities during development.
2. FreeRTOS uses a different approach regarding interrupts. In RTX5 there is no interrupt priority limit for calling non-blocking API functions. The system startup is described in section [Theory of Operation](http://www.keil.com/pack/doc/CMSIS/RTOS2/html/theory_of_operation.html). Its crucial to have the  timer and svc interrupts on the lowest priorities. This is assured by osKernelStart. You should not change the RTOS-related interrupts afterwards. Your softIRQ should have a slightly higher priority to assure it cannot be preempted by timer/svc.
3. Okay, so its sounds link the scheduler has stopped working. Can you please try to debug this further? Is the idle thread still running? Or is core1 trapped in an endless loop?
4. Please do so. RTX5 for Cortex-A is not widely used, yet. Your feedback is extremely beneficial.
5. That sounds like the scheduler is affected as a whole. I am curious to know what happens.

Best,
Jonatan

jameszxj, 2017-12-12:
Hi @JonatanAntoni 
I do some more tests and the results is:
1.the idle thread is still running
2.the OS tick is stilling running
3.no exceptions happened
4.no message from osRtxErrorNotify
It seems that release semaphore is blocked.

Would you give me some more advices，I am not familiar with RTX2 core.

jameszxj, 2017-12-13:
Hi @JonatanAntoni
I do some more tests and the results is:
1.the idle thread is still running
2.the OS tick is stilling running
3.no exceptions happened
4.no message from osRtxErrorNotify

Would you give me some more advices，I am not familiar with RTX2 core.

JonatanAntoni, 2017-12-13:
Hi @jameszxj,

That sounds like the task is not getting unblocked.

If you are using MDK/uVision as IDE you can use builtin tools like [Component Viewer](http://www.keil.com/support/man/docs/uv4/uv4_db_dbg_scvd_viewer.htm) and [Event Recorder](http://www.keil.com/support/man/docs/uv4/uv4_db_dbg_evr.htm) to get some more details about RTX5 state.

With any other IDE not having builtin support for RTX5-awareness you need to dig through the control blocks manually. Watch out for `osRtxInfo` which contains all the wait lists, see `rtx_os.h` for details.

You can try to step into the scheduler as well. Is it possible to halt the program execution with a breakpoint inside the core1 soft-irq routine? You should see the task awaiting the mutex/semaphore/event is unblocked. The actual rescheduling is deferred by setting a post-processing flag `IRQ_PendSV` and finally executed by `osRtxPendSV_Handler` .

You might want to provide a reproducer example if its feasible.

Best,
Jonatan

JonatanAntoni, 2018-01-08:
Hi @jameszxj,

Anything new to this issue? Did my latest hints help to figure out the root cause?

Best,
Jonatan

JonatanAntoni, 2018-02-06:
Closing this issue. Please feel free to reopen if further discussion is needed.

closed 2018-02-06 by JonatanAntoni