ARM-software/CMSIS_5 #290: ITM_TPR_PRIVMASK_Msk size regarding datasheet

Kochise, 2017-12-05:
In the 'CMSIS/Core/Include/core_cmx.h' files, the 'ITM_TPR_PRIVMASK_Msk' is defined as '(0xFUL /*<< ITM_TPR_PRIVMASK_Pos*/)'. However, in the 'armv7m_Architecture_Reference_Manual.pdf', page 775/916 (C.1.7.5) it is stated to be 32 bits wide, while the mask definition limits it to 4 bits.

Which one is right ?

Btw, there are several other deviations from the reference manual. Which one is to follow trustfully ?

JonatanAntoni, 2017-12-05:
Hi @Kochise,

Looks like a deviation from the early days. Thanks for making us aware of it. The reference manuals should be correct but not guaranteed to have no bugs as well ;)

I'll fix the ITM_TPR_PRIVMASK_Msk defines, soon.

Which other deviations do you have spotted? Please let us know.
Pull requests proposing fixes are always appreciated.

Thanks again,
Jonatan

Kochise, 2017-12-05:
There are some missing details/descriptions of some other registers, especially regarding the ITM, like LAR and PIDx/CIDx contents.

Or the TPI_ACPR_PRESCALER_Msk defined on 13 bits by (0x1FFFUL /*<< TPI_ACPR_PRESCALER_Pos*/) (note the pos is commented out) while the datasheet name it TPIU SWOSCALER, page 812/916, and state it is on 16 bits. Not found other ACPR register in the datasheet, so they might correspond to each other.

Someone else will report you additional requests soon.

Btw I previously reported many problem about ST chips as well (ie. https://community.st.com/thread/30924 )

JonatanAntoni, 2017-12-05:
Please find a fix for this issue in 746d9d8.

JonatanAntoni, 2017-12-05:
It's my pleasure.
May I ask you to close this issue if you don't have further concerns, please?

Thanks for contributing,
Jonatan

closed 2017-12-05 by Kochise