ARM-software/CMSIS_5 #291: Debug sequences inadequate for writing flashloaders/handlers

dvasaru, 2017-12-05:
I've been looking at implementing flash loaders using debug sequences. 

Q1. To get reasonable performance out of the loader, we'd need a couple more primitives: Q1A: Read/WriteBLOCK , and Q1B: PollBits. The first one is self explanatory, the second one would save USB roundtrips when polling for a ready or error bit.
Q2: Also missing are definitions for the erase/write sequences including chip erase . Something like "FlashProgramPage" "FlashEraseRow" .
Q3: Is it possible to give arguments to a debug sequence ? Page address and payload would be required for page writes.

JonatanAntoni, 2017-12-06:
Hi @dvasaru,

Thanks for sharing your ideas and questions with us.

Currently the Debug Sequences are not intended to offer Flash Loading capabilities. For this purpose we have the Flash Algorithms in place. These are typically loaded to device RAM and executed from RAM. If this procedure is not applicable for your devices we need to think about alternatives. For sure extending the Debug Sequences would be an option. 

We need to carefully asses what is actually needed in order to develop the Debug Description language further. For the time being the Debug Sequences are executed on the host. If you want to save USB round-trips the Debugger firmware needs to be involved somehow. Thus CMSIS-DAP must be taken into account as well.

We will work on this topic over the next month. Let us know if you come across further requirements.

Thanks,
Jonatan

dvasaru, 2017-12-06:
Hi Jonatan
One additional requirement : on the device I am working on, a CRC may need to be computed as part of page programming . So the language should have a way of efficient indexing into arrays, or a reasonable set of cryptographic primitives.
An "easy" way to achieve some of these goals could be the addition of a blob data type, that could be passed from the debugger (blob of flash page data) through the script to a Blob read/write primitive.  A blob indexing primitive could offer access to individual bytes and could accommodate the CRC computation.

dvasaru, 2018-02-23:
Hi @jonatan
Have you had time to work on this ? We have a working solution based on
named buffers and new buffer DSQ primitives, but it would be good to align
with you.
Dan

JonatanAntoni, 2018-02-23:
Hi Dan,

I cannot tell you about reasonable progress. This issue is still marked as "future" so I have no actual schedule to have anything new.

If you don't mind please add your solution proposal here. Alternatively you can reach out to us via mail.

Your contribution is highly appreciated.

Cheers,
Jonatan

dvasaru, 2018-02-26:
Hi Jonatan

Our solution sounds like this :

We introduced the concept of script buffers, which are byte arrays
maintained outside the script engine (and thus script language independent).
This eliminates the need for array concepts inside the script language
itself, and greatly simplifies parameter marshaling to the script domain.
The script buffer contents are manipulated by a new set of primitives
similar to Write32/Read32.

The following example shows that a single new DSQ primitive, WriteBuffer,
is enough to implement a script based flash loader. The script itself can
run on the host or on a debugger tool.
Yes, it is slower than an on-target flash loader, but it enables flash
writes on devices where execution out of RAM is prohibited.

Flash loader example :
When writing to target flash, the IDE first populates the script buffers,
and then passes their ID to the flash programming script.
In our proof of concept, the payload buffer is at ID 0.  In DSQ spec, this
could be a new variable __flashbuffer, similar to __dp or __ap.

To write from buffers to target, we define a new primitive:

WriteBuffer(unsigned * bufferId*, unsigned *bufferOffset*, unsigned
*address*, unsigned *len*)
---Writes *len* bytes from offset *bufferOffset*  in buffer * bufferId* to
target memory at *address.*

In pseudocode, the IDE does :
void programFlashMemoryOnTarget(unsigned targetAddress,char*flashData, int
len)
{
  Buffer b=bufferManager.CreateBuffer(0,len); // id 0
  b.copyDataFrom(flashData,len); // populate buffer with data to go in flash
  scriptEngine.setVariable("__flashbuffer",0);
  scriptEngine.setVariable("addr", targetAddress );
  scriptEngine.setVariable("len", len);
  invokeScript(programFlash);
}

The implementation of WriteBuffer uses *RDDI_DAP_WriteARMMem* to copy the
specified data subset to target memory.

WriteBuffer(unsigned * bufferId*, unsigned *bufferOffset*, unsigned
*address*, unsigned *len*)
{
  Buffer b=bufferManager.GetBuffer(bufferId);
  ByteArray ba=b.getSubset(bufferOffset,len);
  RDDI_DAP_WriteARMMem(ba,address,len);
}

The script itself chunks the payload into individual pages if needed, and
accesses the flash controller SFRs for the actual erase/write:

for (each flash page in len bytes)
  WriteBuffer(__flashbuffer, currentOffset, flashControllerPageBufferAddress,
pageSize)
  Write32 .... # tell flash controller to erase and write the page
  currentOffset += pageSize
  #adjust flashControllerPageBufferAddress if needed

So the proposal is to add __flashbuffer as a variable and Read/WriteBuffer
primitives to the DSQ spec.
This is work in progress for us so we will probably add more buffer
primitives along the road (crc/crypto)

Your comments are welcome.

Best regards,
Dan