ARM-software/CMSIS_5 #293: Calculate the phase of a complex number

escherstair, 2017-12-14:
CMSIS-DSP library offers some functions to calculate the magnitude of a complex number (i.e. arm_cmplx_mag_q31(), ...).

Is there a function to calculate the phase too?
Or should I write one by myself?

llefaucheur, 2017-12-14:
Hi, the number of APIs with complex numbers will grow. Give us one or two months to consolidate the requirements from different sources. By the way, this is time to tell, so what would you be interest ? (which FFT length, dynamic range, ..).

escherstair, 2017-12-14:
Hi,
in this moment I'm interested in a basic function (I mean, that works on a single complex number).
I need to convert a complex number from cartesian (real + imaginary) to polar (magnitude + phase) coordinates.
More in general, I'm interested in functions to handle FFT with lenght from 512 to 8192 samples in q31_t format.

Even functions to calculate "cross power spectrum" (or "[Cross-spectral density](https://en.wikipedia.org/wiki/Spectral_density#Cross-spectral_density)") would be useful.

llefaucheur, 2017-12-14:
Yes, we intend to provide efficient kernels for beamformers as an example. What would you be interested from [vDSP](https://developer.apple.com/documentation/accelerate/vdsp) ?

escherstair, 2018-03-19:
Sorry for my late answer but I completely missed your last request.
This is a list of functions I would be interested in:

- Single-Vector Polar-Rectangular Conversion
- Single-Vector Complex Conjugation
- Single-Vector Integration
- Vector-to-Vector Spectra Computation
- Vector-to-Vector Coherence Function Computation
- Vector-to-Vector Transfer Function Computation
- Windowing

But many of the functions from vDSP are useful in some contexts.

llefaucheur, 2018-03-20:
All the above are in the to-do list. 
Just for alignment 
Single-Vector Integration : will the sum of all absolute elements DASUM(BLAS)/F06EKF(NAG).
Vector-to-Vector Spectra Computation will be DOTC and DOTU
Vector-to-Vector Coherence Function Computation  will be : X.*conj(Y)/abs(X).*abs(Y)
Vector-to-Vector Transfer Function Computation will be : X./Y
Windowing : chebwin, blackmanHarris, hamming, hanning

escherstair, 2018-03-20:
> Single-Vector Integration : will the sum of all absolute elements DASUM(BLAS)/F06EKF(NAG).
I think this should implement a digitl integrator using one of the following methods:

- rectangular integrator
- trapezoidal integrator
- Simpson's rule

tachen-cs, 2020-07-13:
Hi, do we have such a function now?
 \ie, computing phase values after we getting complex numbers from a function like 
arm_rfft_q31(&_S31, (q31_t*)_sampleBuffer, (q31_t*)_fftBuffer);

Or, can I simply implement such a function like:

for (n = 0; n < numSamples; n++) {
    pDst[n] = atan2(pSrc[(2*n)+0] + pSrc[(2*n)+1]);
}

christophe0606, 2020-07-15:
@tachen-cs Arg function is not available.
If you want to use atan2, you'll need to convert back the q31 to float (those conversion functions are available).

tachen-cs, 2020-07-15:
> @tachen-cs Arg function is not available.
> If you want to use atan2, you'll need to convert back the q31 to float (those conversion functions are available).

Thanks for your reply!
Would you add such function recently as phase is another important feature in frequency domain?

christophe0606, 2020-07-16:
I can't communicate any schedule.

tachen-cs, 2020-08-11:
> Hi, do we have such a function now?
> \ie, computing phase values after we getting complex numbers from a function like
> arm_rfft_q31(&_S31, (q31_t*)_sampleBuffer, (q31_t*)_fftBuffer);
>
> Or, can I [...]

Dear Christophe @christophe0606,
I implement the above phase function under float data type, it seems OK (I would test it further use my algorithm) as the range is always btw -3.14 to 3.14.  Thanks for your help.

I still have a question.  When I read the source code of _arm_cmplx_mag_f32.c_, I found different codes under different macros (and they seems to do things which I don't know, \eg,  _inlined Fast SQRT using inverse SQRT newton-raphson method_, it seems to speed the mag function up? I am not sure am I right?  And I just choose the most stupid way similar to the **last part** in this function to implement the phase function).

I write the codes on **cortex mo+ board**, and I am not sure which macro I should follow like the mag function codes, or my current implementation is OK, may you give my some hints?

christophe0606, 2020-08-12:
@tachen-cs  On Cortex-M0+, choosing the last part is the right way.
Other methods are used on Cortex-M with DSP extensions which is not the case of M0+

FlixofTheWrist, 2021-08-10:
Hey, I have a question on the same subject: calculating the phase of a chosen frequency of the arm_cfft_f32() function. I will start by telling you about my application and code.

The input buffer is 8192 samples of a 60Hz perfect sin signal. (for each real number, I associate a 0)
The sample speed is 20 480 Hz.
Since sampling speed is 20 480Hz and buffer of the output is  4096, my frequency precision is 5Hz. (5*12 = 60Hz)
So in the Input buffer, after computing the CFFT, bin of fundamental should be 24 and 25. (60Hz real and imaginary).
Input analog signal is generated by Analog Discovery 2 and is a sin wave that is centered at 1.67V with amplitude of 1.67V.
The phase of the analog sin wave is set at 0 on the software provided (waveform) by the Analog Discovery 2.

[image]

I am building a code that can find the harmonics on a 60Hz experimental power grid. 
The function arm_cmplx_mag_f32() works just fine to find the magnitude of detected harmonics. Good precision.
The problem is with the phase. It is important to know the phase to be able to apply the right correction on the given harmonic to eliminate it completly.

So, i tried using the atanf2 function to find the phase of a given real and imaginary pair.

However, the problem is that after computing the CFFT multiple times, the value are never the same for the SAME sin wave.

In other words, the function atanf2 can't find an accurate value for the phase since it's never the same.

Do you guys know why i can't calculate the phase with precision but I can find the magnitude easily.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606

NeiD52, 2023-01-31:
> Здравствуйте, есть ли у нас сейчас такая функция? \ т. е. расчет фаз после получения комплексных чисел из таких функций, как arm_rfft_q31(&_S31, (q31_t*)_sampleBuffer, (q31_t*)_fftBuffer);
> 
> Или я просто могу реализовать такую ​​функцию, как:
> 
> for (n = 0; n < numSamples; n++) { pDst[n] = atan2(pSrc[(2 _n)+0] + pSrc[(2_ n)+1]); }

hi! could you show your code how to calculate the phase?

llefaucheur, 2023-01-31:
Hi ATAN2 is here : https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FastMathFunctions/arm_atan2_f32.c

NeiD52, 2023-01-31:
> Hi ATAN2 is here : https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FastMathFunctions/arm_atan2_f32.c

thanks for the feature link! I understand correctly what it is y/x, y - real and x - imag?

llefaucheur, 2023-01-31:
We usually align with Matlab : https://www.mathworks.com/help/matlab/ref/atan2.html