ARM-software/CMSIS_5 #294: Trigonometric functions in CMSIS-DSP

escherstair, 2017-12-14:
CMSIS-DSP offers only few trigonometric functions (`arm_sin`, `arm_cos` and `arm_sin_cos`, as far as I can see).
Some other functions are useful for my needs; I mean:

- [tangent](https://en.wikipedia.org/wiki/Trigonometric_functions#tangent) - I know I can calculate it using `arm_sin` and `arm_cos`

- [atan2](https://en.wikipedia.org/wiki/Atan2)

- `arcsin `and `arccos`

For the end developer it's quite difficult to implement the inverse trigonometric functions.
I'm almost interested in q31_t format.

Do you think they would be available in a future release?

llefaucheur, 2017-12-14:
Hi, among the various implementation options we can propose an algorithm optimised for speed using one of the series of the "Handbook of Mathematical Functions" (Abramowitz & Stegun).

escherstair, 2017-12-15:
Hi,
do you mean CMSIS-DSP library will be extended with this kind of functions?
Or are you suggesting me to implement them following the "Handbook of Mathematical Functions"?

llefaucheur, 2017-12-15:
Hi, We will implement the circular and inverse circular functions using formulas of this book. Regards.

pliniofpa, 2018-07-13:
Laurent,
With the answer above you are saying that CMSIS doesn't have plans to extende the math library to include atan, asin, acos, atan2, etc?

llefaucheur, 2018-07-15:
Hi pliniofpa, we will extend CMSIS_DSP in several directions, including mathematical functions. My point with escherstair, is to let you propose an algorithm based on public litterature.

DanSzaboStoneAge, 2021-04-27:
FWIW, I just realized that atan2 wasn't supported (oops), and ended up using the stdlib version.  I found [this article](https://www.dsprelated.com/showarticle/1052.php) which has a nifty approximation of atan2 with some further refinements in the comments, all of which looks pretty speedy.  It'd be interesting to compare with the LUT based atan implementation that seems to in progress.  Cheers.

christophe0606, 2021-04-28:
@DanSzaboStoneAge Thanks for feedback. Yes, Remez is a good method for finding approximations.
LUT can be ok too if the table is not too big.

DanSzaboStoneAge, 2021-04-28:
@christophe0606 De nada.  The Remez bit is just for finding coefficients, the actual atan() function uses a polynomial, and a rational expression is used for comparison, which seems about as quick as you could hope for.  It looks like the LUT in #1164 is set by `[FAST_MATH_TABLE_SIZE + 2]`, which I reckon would put it at 514.  Not sure if that's too big or not.  Thanks much.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606