ARM-software/CMSIS_5 #296: RTX5.20 LPC43xx M0 Alternate timer

Concrett, 2017-12-18:
Hi,

I encounter an issue when updating CMSIS 5.01 to CMSIS 5.2
I am using Keil to compile and Flash a LPC4337 processor.
Both core are using RTX5, M4 core is using Systick timer for the kernel and M0 core is using the RIT Timer as an alternate timer.

With previous version (CMSIS 5.01) I used this code to configure my alternate timer : 

```
////
#include "rtx_os.h"                     // ARM::CMSIS:RTOS2:Keil RTX5
#include "LPC43xx.h"                    // Device header

/*
** forward the M0_SVC_Handler to the RTX SVC_Handler
*/
__asm __declspec( noreturn ) void M0_SVC_Handler( void )
{
  EXTERN SVC_Handler
  LDR R0, =SVC_Handler
  BX R0
}

/*
** forward the M0_PendSV_Handler to the RTX PendSV_Handler
*/
__asm __declspec( noreturn ) void M0_PendSV_Handler( void )
{
  EXTERN PendSV_Handler
  LDR R0, =PendSV_Handler
  BX R0
}

/*
** forward the M0_RIT_OR_WWDT_IRQHandler to the RTX SysTick_Handler
*/
__asm __declspec( noreturn ) void M0_RITIMER_OR_WWDT_IRQHandler( void )
{
  EXTERN SysTick_Handler
  LDR R0, =SysTick_Handler
  BX R0
}

int32_t osRtxSysTimerSetup( void )
{
  LPC_CCU1->CLK_M4_RITIMER_CFG = CCU1_CLK_M4_RITIMER_CFG_RUN_Msk ;
  LPC_RITIMER->COUNTER = 0x00000000 ;
  LPC_RITIMER->MASK = 0x00000000 ;
  LPC_RITIMER->COMPVAL = osRtxInfo.kernel.sys_freq / osRtxConfig.tick_freq - 1 ;
  
  return( M0_RITIMER_OR_WWDT_IRQn ) ;  /* Return IRQ number of SysTick timer */
}

void osRtxSysTimerEnable( void )
{
  LPC_RITIMER->CTRL = RITIMER_CTRL_RITENBR_Msk | RITIMER_CTRL_RITENCLR_Msk | RITIMER_CTRL_RITEN_Msk ;
}

void osRtxSysTimerDisable( void )
{
  LPC_RITIMER->CTRL = 0 ;
}

void osRtxSysTimerAckIRQ( void )
{
  /* Acknowledge timer interrupt. */
  LPC_RITIMER->CTRL |= ( RITIMER_CTRL_RITINT_Msk ) ;
}

uint32_t osRtxSysTimerGetCount( void )
{
  uint32_t tick;
  uint32_t val;

  tick = ( uint32_t )osRtxInfo.kernel.tick ;
  val = LPC_RITIMER->COUNTER ;
  if( LPC_RITIMER->CTRL & ( RITIMER_CTRL_RITINT_Msk ) )
  {
    val = LPC_RITIMER->COUNTER ;
    tick++ ;
  }
  val += tick * ( LPC_RITIMER->COMPVAL + 1 ) ;
  
  return( val ) ;
}

uint32_t osRtxSysTimerGetFreq( void )
{
  return( osRtxInfo.kernel.sys_freq ) ;
}
////
```

With the new CMSIS versions (CMSIS 5.10 and CMSIS 5.20) the code does not compile anymore.
The variable "osRtxInfo.kernel.sys_freq" does not exist in these versions (I assume that I simply need to replace it by SystemCoreClock variable).

With this modification my project compiles but as soon as I call "osDelay" or "osDelayUntil" functions, my code gets stuck. I am not able to use Delay functions anymore (nothing else was changed in my code and if I re-use CMSIS 5.01 everything works fine).

What should I add/modify to be able to update the CMSIS software pack from 5.01 to 5.2 and keep using the RIT Timer as alternate systick timer for my LPC4337 M0 core (I have no issue with the LPC4337 M4 core, which uses the systick timer) when updating the CMSIS software pack?

I tried to contact Keil Support but they assumed that it should be caused by internal API changes so maybe someone here could help me !

Regards.

RobertRostohar, 2017-12-18:
RTX V5.2.0 tick is based on a new generic OS Tick API which should simplify external timer implementations.

OS Tick API documentation:
http://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__TickAPI.html

SysTick implementation can be used as a reference and can be found here:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/Source/os_systick.c

Your existing timer implementation using osRtxSysTimerXxxx functions should be ported to the mentioned OS Tick which should be straightforward.

Concrett, 2017-12-18:
Thank you for your quick reply.
Indeed with minor modifications using SysTick implementation as reference everything works fine !

Regards.

closed 2017-12-18 by Concrett