ARM-software/CMSIS_5 #301: __set_MVBAR, __get_MVBAR, __set_VBAR, __get_VBAR

ua1arn, 2018-01-02:
__set_VBAR use __set_CP(15, 0, vbar, 12, 0, 1) - this opcode for Monitor Vector Base Address Register

Vector base address should be accessed trough __set_CP(15, 0, vbar, 12, 0, 0)

Proposed code:

 /** \brief  Get MVBAR

    This function returns the value of the Monitor Vector Base Address Register.

    \return               Monitor Vector Base Address Register
 */
__STATIC_FORCEINLINE uint32_t __get_MVBAR(void)
{
  uint32_t result;
  __get_CP(15, 0, result, 12, 0, 1);
  return result;
}

/** \brief  Set MVBAR

    This function assigns the given value to the Monitor Vector Base Address Register.

    \param [in]    mvbar  Monitor Vector Base Address Register value to set
 */
__STATIC_FORCEINLINE void __set_MVBAR(uint32_t vbar)
{
  __set_CP(15, 0, vbar, 12, 0, 1);
}

 /** \brief  Get VBAR

    This function returns the value of the Vector Base Address Register.

    \return               Vector Base Address Register
 */
__STATIC_FORCEINLINE uint32_t __get_VBAR(void)
{
  uint32_t result;
  __get_CP(15, 0, result, 12, 0, 0);
  return result;
}

/** \brief  Set VBAR

    This function assigns the given value to the Vector Base Address Register.

    \param [in]    vbar  Vector Base Address Register value to set
 */
__STATIC_FORCEINLINE void __set_VBAR(uint32_t vbar)
{
  __set_CP(15, 0, vbar, 12, 0, 0);
}

JonatanAntoni, 2018-01-08:
Hi @ua1arn,

Thanks for reporting. May I ask you to double check the fix I added with 0ac0276, please?

Best,
Jonatan

ua1arn, 2018-01-08:
In part of function bodies looks fine.

JonatanAntoni, 2018-01-09:
Issue should be solved. Please feel free to reopen if any questions left.

closed 2018-01-09 by JonatanAntoni

ua1arn, 2018-01-09:
Tested on Renesas R7S721020VCFP#AA0 (standalone application) - work as expected.

JonatanAntoni, 2018-01-09:
Great, many thanks for your additional testing efforts.