ARM-software/CMSIS_5 #302: Why is PRIGROUP preserved when performing SYSRESETREQ

c1728p9, 2018-01-02:
When performing a software reset with [__NVIC_SystemReset](https://github.com/ARM-software/CMSIS_5/blob/873bb94c9c570773ab6f12587b93e76f56d40193/CMSIS/Core/Include/core_cm7.h#L2141) the PRIGROUP field is reserved. This shouldn't be necessary, as the ARMv7m indicates that this register has a reset value of 0xFA050000. Note - I'm not advocating the removal of this code, just trying to understand it.

Are there cases where a software reset via SYSRESETREQ doesn't actually reset PRIGROUP? Is  [the code to preserve PRIGROUP](https://github.com/ARM-software/CMSIS_5/blob/873bb94c9c570773ab6f12587b93e76f56d40193/CMSIS/Core/Include/core_cm7.h#L2141) added to workaround some known errata or bug?

Related issue in DAPLink here:
https://github.com/ARMmbed/DAPLink/pull/362

JonatanAntoni, 2018-01-09:
Hi @c1728p9,

changing PRIGROUP while the system is running may have serious side effects. E.g. if you are running RTX5 the RTOS-related interrupt priorities are calculated with respect to the PRIGROUP. Changing PRIGROUP while RTX5 is running can instantly crash the system. Thus we typically use read-before-write semantics to assure we change only the relevant bits and prevent unintended side effects.

Unfortunately I cannot describe why clearing PRIGROUP may actually be a problem in case of DAPLink on i.MXRT (Cortex-M7). In addition the problem description is very unspecific.

Cheers,
Jonatan

c1728p9, 2018-01-25:
Thanks makes sense. Thanks for the info.

closed 2018-01-25 by c1728p9