ARM-software/CMSIS_5 #306: IAR <intrinsics.h> conflicts with CMSIS file "cmsis_iccarm.h"

quantum-leaps, 2018-01-16:
The new CMIS header file "cmsis_iccarm.h" reports an error when the IAR <intrinsics.h> is included independently from CMSIS:

` #ifdef __INTRINSICS_INCLUDED
  #error intrinsics.h is already included previously!
  #endif
`
This is undesirable for system-level code that wants to be independent from CMSIS (capable of working with or without CMSIS). In the interest of introducing minimal dependency, such code needs to include only <intrinsics.h> and not the whole "cmsis_iccarm.h". But then, such system-level code is incompatible with CMSIS.

quantum-leaps, 2018-01-17:
Actually, the new header file `cmsis_iccarm.h` breaks many existing software, such as STM32Cube, FreeRTOS ports and most likely many others. The software simply no longer compiles. This renders this CMSIS version essentially unusable for the IAR EWARM users.

JonatanAntoni, 2018-01-17:
Hi @quantum-leaps,

that sounds like a mistake happened during transfer of cmsis_iar.h to cmsis_iccarm.h. Sorry for any inconvenience arising from this. I already made IAR aware and hopefully we can come up with a solution that's recreates backward compatibility.

Best,
Jonatan

TTornblom, 2018-01-18:
Can you provide an example project showing this issue?

I've just configured and built an example STM32Cube project where I include <intrinsics.h> in main.c and there is no problem compiling this project, but I guess the project does not include cmsis_iccarm.h.

Including <intrinsics.h> after cmsis_iccarm.h will also not cause a problem.

Thanks,
Thomas

JonatanAntoni, 2018-06-22:
Hi @quantum-leaps,

Did you recognize the question from @TTornblom? He is working with IAR and maintaining the `cmsis_iccarm.h` file. Can you give us some more details about the include chain that causes your issue?

Thanks,
Jonatan

kjetilos, 2018-06-25:
Hi @JonatanAntoni 

I see the same issues in both FreeRTOS and Micrium OS applications. The problem is that with the cmsis_iccarm.h header file you are forcing the user/application to manually control the header file include order.

You can reproduce this issue by building the FreeRTOS demo for the SLSTK3401A kit for instance. In this FreeRTOS demo the FreeRTOS header files are included first and they will bring in the porting layer header files from the generic Cortex-M4 port of FreeRTOS. These header files will include <intrinsics.h>. After the FreeRTOS headers are included, the CMSIS Device header files for the device is included in the application and that will eventually pull inn cmcis_iccarm.h which will trow a compiler error because intrinsics has already been included.

Kjetil

JonatanAntoni, 2018-06-25:
Hi @kjetilos,

I guess you refer to the forced compilation error at [cmsis_iccarm.h:483](https://github.com/ARM-software/CMSIS_5/blob/d04f55fbaa19db50eea61639563107060bee63cf/CMSIS/Core/Include/cmsis_iccarm.h#L483), i.e. having intrinsics.h included before cmsis_iccar.h. Perhaps @TTornblom can justify this implementation? 

It looks like some of the intrinsic functions defined in that header are not compatible to CMSIS. Hence they are overwritten, like `__CLZ` on Cortex-M0 or `FPSCR` on devices with FPU. This in turn leads to a dependency on the include order (not to be mixed up with include path order).

I am not yet sure how to deal with that correctly. CMSIS aims to offer a generic, device and compiler independent interface. Including compiler specific headers manually seems to bypass the compiler independence. Why do you need to include intrinsics.h file manually at all?

In case of FreeRTOS we have a conflict between multiple compiler abstractions, yes. Effectively FreeRTOS.h includes portable.h which includes the compiler (i.e. port) specific portmacro.h which (IAR port) finally includes intrinsics.h. With our [CMSIS repackaging of FreeRTOS](https://github.com/ARM-software/CMSIS-FreeRTOS) we worked around that problem. We assure that the CMSIS-Core headers are included using the [FreeRTOSConfig.h](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/cbb8f4789730477f1dc32bcbf5b3310096787c04/CMSIS/RTOS2/FreeRTOS/Config/ARMCM/FreeRTOSConfig.h#L46) before the `portable.h`. I admit this is not a perfect solution.

Do you have proposals how to get this improved?

Cheers,
Jonatan

TTornblom, 2018-06-25:
I've been looking into this issue, and it and the explanation I got from one of the developers is that intrinsics.h clashes with some parts of CMSIS, which is why cmsis_iccarm.h, as of EWARM 8.20, instead includes iccarm_builtin.h. Apparently intrinsics.h predates the parts of CMSIS that causes the issues, so it can not safely be fixed to avoid the issues.

In EWARM 8.20 (and later) intrinsics.h includes iccarm_builtin.h and basically defines all the functions of intrinsics.h using the builtins in that file.

What version of EWARM is causing these issues? My guess is 8.11 (or older) as the test and error message should not be used on 8.20 (or later).

The discriminator is the symbol "__ICCARM_INTRINSICS_VERSION__", which is 0 for 8.11 and older and 2 for 8.20 and newer.

Cheers,
Thomas

kjetilos, 2018-06-25:
Hi @TTornblom 

I am using EWARM 7.80.4, I can try reproduce this on a newer version. Does this mean that in effect CMSIS 5 requires EWARM 8.20 or later?

JonatanAntoni, 2018-06-25:
Hi @kjetilos,

CMSIS of course aims to support a wide set of compilers and versions. Admittedly official support is limited to recent versions of compilers and tools. We expect that most users updating CMSIS should be able to update the compiler as well. But that should not cause problems for older tools which could be seen as under community support, e.g. you ;)

In your concrete case the clash is caused by the compatibility layer for older compiler versions as @TTornblom stated. You should be able to build your application without issues if its feasible for you to switch to a recent version of the compiler.

In any case your feedback is very appreciated. Improvement proposals are always valuable.

Cheers,
Jonatan

kjetilos, 2018-06-25:
Hi, when testing with EWARM 8.22.4 I don't see the issue. I hope that there can be a fix that allows EWARM 7.80.4 to be supported as well. What is the effects of potentially removing/ignoring this forced compilation error message?

Kjetil

JonatanAntoni, 2018-06-25:
@kjetilos,

you could give it a try. I expect the compiler failing elsewhere because of redefined functions.

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
Closing due to inactivity.

closed 2019-03-12 by JonatanAntoni