ARM-software/CMSIS_5 #311: Missing declaration of structures for real-value inputs in arm_const_structs.h

ballotz, 2018-02-01:
In arm_const_struct.h there are all cfft declarations `extern const arm_cfft_instance_` but are missing all the `extern const arm_rfft_fast_instance_` structures.

in arm_const_struct.c
```
const arm_rfft_fast_instance_f32 arm_rfft_fast_sR_f32_len32 = {
	{ 16, twiddleCoef_32, armBitRevIndexTable32, ARMBITREVINDEXTABLE_16_TABLE_LENGTH },
	32u,
	(float32_t *)twiddleCoef_rfft_32
};
```
I think `twiddleCoef_32, armBitRevIndexTable32` should be `twiddleCoef_16, armBitRevIndexTable16`.
Original version causes hard-fault in my project.

llefaucheur, 2018-02-02:
Hi, you are true. The mix 16/32 in the line "..Table32, ..TABLE_16_.." should have triggered our eyes. 
During the review I also found this issue "Sint->fftLen = S->fftLenRFFT / 2;" where a data is load to a "const" memory. Which is an other source of issue if your tables are in flash.
I am now trying to understand why the issues passed through the non-regression test framework.

christophe0606, 2019-05-29:
It looks like those structures are not used since arm_rfft_fast_init_f32 is basically redefining the same things.
And arm_rfft_fast_init_f32 does not have the bug and use twiddleCoef_16, armBitRevIndexTable16.

closed 2019-05-29 by christophe0606

Bambofy, 2019-11-02:
So how do we use arm_rfft_fast_f32()?

christophe0606, 2019-11-05:
@Bambofy Just use arm_rfft_fast_init_f32 to init your structure.
Then you can reuse it with arm_rfft_fast_f32 (some casting may be required because of the const attribute).

I agree it is a bug. Because in arm_rfft_fast_f32, the structure is changed to compute the field Sint->fftLen but this has already been done with arm_rfft_fast_init_f32.

And this field is also already initialized in the const structure.

arm_rfft_fast_f32 should not do this change and be able to work with a const structure.

So, I open a bug to track this issue.

ballotz, 2019-11-05:
I agree that arm_rfft_fast_f32 should be able to work with a const structure.
The problem with arm_rfft_fast_init_f32 is also that the dynamic switch inside prevents the linker from stripping unused tables, even if you know that you are only using one (or some, likely small) FFT size in your code. This is why I initialized arm_rfft_fast_instance_f32 by my own referencing only one specific table.

christophe0606, 2019-11-05:
@ballotz There are now some compilation flags to keep only the table you need … but you have to rebuild the library.