ARM-software/CMSIS_5 #312: arm_calc_2pow() in various math_helper examples is needlessly inefficient.

WestfW, 2018-02-01:
Various examples include a file "math_helper.c" that has code to calculate powers of 2:

```
/**
 * @brief  Calculates pow(2, numShifts)
 * @param[in]  numShifts 	number of shifts
 * @return pow(2, numShifts)
 */
uint32_t arm_calc_2pow(uint32_t numShifts)
{

  uint32_t i, val = 1;

  for (i = 0; i < numShifts; i++)
    {
      val = val * 2;
    }

  return(val);
}
```

Surely a "static inline uint32_t arm_calc_2pow(uint32_t numShifts) { return 1<<numshifts; }" in the .h file would work fine and be efficiently implemented on all current Cortex M processors?

llefaucheur, 2018-02-02:
You are true.. and this subroutine is used in arm_apply_guard_bits(), where the line "pIn[i] = pIn[i] * arm_calc_2pow(guard_bits);" could be rewrote using hacker's style : "pIn[i] <<= guard_bits;"
Fortunately, this code, replicated several times, seems coming from early stages of the project. For information, we took the action the rewrite some pieces of the test framework, and the way nested macros are implemented. By the way, any good proposal is welcomed at this point.

christophe0606, 2020-01-23:
It no more applies to the current version of the code (those functions no more exist).

closed 2020-01-23 by christophe0606