ARM-software/CMSIS_5 #313: Memory pool cannot be allocated for unaligned structs

maldata, 2018-02-02:
I created a Mail<T, size> object where the type T was a struct with size 258 bytes. The program fails to create the Mail<> object.

When the memory pool is allocated in rtx_mempool.c, the size of the object is [rounded up to a multiple of 4](https://github.com/ARM-software/CMSIS_5/blob/bb78a8aa71384f2c7d909570d88d3cce3467ffe1/CMSIS/RTOS2/RTX/Source/rtx_mempool.c#L213). Shortly thereafter, [an error is reported](https://github.com/ARM-software/CMSIS_5/blob/bb78a8aa71384f2c7d909570d88d3cce3467ffe1/CMSIS/RTOS2/RTX/Source/rtx_mempool.c#L246) if the object's size was not a multiple of 4.

I did not dig in any deeper to determine if there is a compelling reason for this limitation, but I wanted to submit an issue to see if this is the intentional behavior.

Thanks

JonatanAntoni, 2018-02-05:
Hi maldata,

can you please double check your call to `osMemoryPoolNew`?

The calculations you refer to shall assure that pointers to pool objects can always be aligned to word boundaries. Thus the size for one object is rounded up. As a consequence the overall pool memory needs to be larger as well. 

The error you get is not about a single object but about the overall pool memory. Do you provide an user defined pool memory through `osMemoryPoolAttr_t::mp_mem`? Please double check alignment and size of this memory. It needs to be properly word-aligned and large enough for `block_count` times the padded Mail<> object.

Best,
Jonatan

maldata, 2018-02-05:
Hi Jonatan,

In this case, I am not explicitly allocating the pool myself. I am using Mbed, which provides a mail queue objects. I am statically creating the Mail<> object, and its constructor allocates the pool using osMemoryPoolNew. My code is very similar to [the example here](https://os.mbed.com/docs/v5.7/reference/mail.html).

I simply define a struct (which, in my case, was **not** word-aligned), and then do something like `Mail<mail_t, 16> mail_box;`

Should this be considered a bug in Mbed's implementation, rather than the underlying CMSIS?

Thanks!

JonatanAntoni, 2018-02-06:
Hi maldata,

That sounds like an issue with the wrapper class template provided by mbed. This issue should have been fixed a while ago, see [5abbccb](https://github.com/ARMmbed/mbed-os/commit/5abbccb45c17de1f07a7f512a6ad832b64ee5396). Please check your version of mbed.

Cheers,
Jonatan

ReinhardKeil, 2018-04-23:
closing as no further activity

closed 2018-04-23 by ReinhardKeil